MagicMirrorOrg/MagicMirror #1739: missing modules

9170895137, 2019-07-26:
I installed the magic mirror but noticed when it got to the end of just be asking to have it auto start and disable screen saver it said library and modules missing.

MichMich, 2019-07-26:
This is the fifth issue you open. As said please use the forum. No support is given via GitHub. Please stop opening issues. Thanks. 

Please refer to http://forum.magicmirror.builders for support and troubleshooting.

closed 2019-07-26 by MichMich

9170895137, 2019-07-26:
I don’t see the issues when I do a search and as for that forum everything is grouped together. Like troubleshoot 1 topic and a lot of threads

From: Michael Teeuw
Sent: Friday, July 26, 2019 5:42 AM
To: MichMich/MagicMirror
Cc: 9170895137; Author
Subject: Re: [MichMich/MagicMirror] missing modules (#1739)

This is the fifth issue you open. As said please use the forum. No support is given via GitHub. Please stop opening issues. Thanks.
Please refer to http://forum.magicmirror.builders for support and troubleshooting.