MagicMirrorOrg/MagicMirror #1747: All Travis Build Fail

MichMich, 2019-07-31:
I noticed all PR's failed during travis. It turned out that the following line in the `before_script` failed:
```
  - sh -e /etc/init.d/xvfb start
```
It gave the following error:
```
sh: 0: Can't open /etc/init.d/xvfb

The command "sh -e /etc/init.d/xvfb start" failed and exited with 127 during .
```
After some [googling](https://stackoverflow.com/questions/55674746/travis-sh-0-cant-open-etc-init-d-xvfb), the solution was to use the following in stead: 
```
services:
  - xvfb
```

This solves the 127 error, but unfortunately, all test that need to spin up electron are now timing out. Failing the travis test. It seems Electron is not able to spin up. 

Suggestions/solutions are welcome.

MichMich, 2019-07-31:
@roramirez Any idea?

roramirez, 2019-08-01:
Let me check something I'll and find what can be wrong with this

roramirez, 2019-08-03:
Fixed in https://github.com/MichMich/MagicMirror/pull/1748

closed 2019-08-03 by MichMich