MagicMirrorOrg/MagicMirror #1751: EADDRINUSE error assistant not working

sidevil, 2019-08-08:
npm start

> magicmirror@2.8.0 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.8.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
No helper found for module: compliments.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
Initializing new module helper ...
Module helper loaded: MMM-AlexaPi
Initializing new module helper ...
Module helper loaded: MMM-NetworkConnection
No helper found for module: MMM-iFrame.
No helper found for module: MMM-page-indicator.
Initializing new module helper ...
Module helper loaded: MMM-MusicOnDemand
Initializing new module helper ...
Module helper loaded: MMM-AssistantMk2
Initializing new module helper ...
Module helper loaded: MMM-Hotword
Initializing new module helper ...
Module helper loaded: MMM-NotificationTrigger
No helper found for module: MMM-pages.
All module helpers loaded.
Starting server on port 8080 ... 
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Connecting socket for: MMM-AlexaPi
Starting node helper for: MMM-AlexaPi
Connecting socket for: MMM-NetworkConnection
MMM-NetworkConnection helper started ...
Connecting socket for: MMM-MusicOnDemand
Connecting socket for: MMM-AssistantMk2
Connecting socket for: MMM-Hotword
[HOTWORD] MMM-Hotword starts
Connecting socket for: MMM-NotificationTrigger
Sockets connected & modules started ...
Whoops! There was an uncaught exception...
{ Error: listen EADDRINUSE 0.0.0.0:8080
    at Server.setupListenHandle [as _listen2] (net.js:1330:14)
    at listenInCluster (net.js:1378:12)
    at doListen (net.js:1492:7)
    at process._tickCallback (internal/process/next_tick.js:63:19)
    at Function.Module.runMain (internal/modules/cjs/loader.js:755:11)
    at startup (internal/bootstrap/node.js:241:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:579:3)
  errno: 'EADDRINUSE',
  code: 'EADDRINUSE',
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8080 }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.
n^CShutting down server...
Stopping module helper: updatenotification
Stopping module helper: calendar
Stopping module helper: newsfeed
Stopping module helper: MMM-AlexaPi
Stopping module helper: MMM-NetworkConnection

MichMich, 2019-08-08:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-08-08 by MichMich

sidevil, 2019-08-08:
ok :) i will post on forum