MagicMirrorOrg/MagicMirror #1758: if user runs npm install in module folder with no package.json, node_helper folder is erased

sdetweil, 2019-08-29:
if you run npm install in  a module folder with no package.json, a side effect is that the MM shipped modules/node_modules/node_helper/index.js is erased, and now nothing (with node helper.js) will run, and MM goes black screen.. 

without knowledge of git, github, etc, there is no recovery  of this required module..

happens on 2.8.0
on my odroid its node 8.10.0, npm 6.10.2, unbuntu 18.04.3
on pi its node 10.6.0, npm 6.1.0,  Raspbian 9.4/stretch
on jetson nano, its node 10.16.3, npm 6.9.0, ubuntu 18.04.2

i think we might have to specify this as a dependency, and have a github repo for it.. 
https://stackoverflow.com/questions/28728665/how-to-use-private-github-repo-as-npm-dependency/28729646#28729646

esnalabu, 2019-09-02:
Can confirm. Leaves my MM all black.

A workaround is to checkout the `modules/node_modules/node_helper/index.js` file from the git repo after each npm install:

`git check modules/node_modules/node_helper/index.js`

sdetweil, 2019-09-02:
right.. (should be checkout , not check) but if you don't know git that well, etc.. its a fun time..
i think the node helper module should be a repo on michmich github and then referenced in package.json. then this won't happen

  "node_helper": "git://github.com/MichMich/node_helper.git",

lavolp3, 2019-09-04:
why does this even happen? Shouldn't an `npm install` in a module folder be local and limited to that folder/repo?
Also, this hasn't happened before 2.8.0 has it?

sdetweil, 2019-09-04:
I hadn't seen reports of this until recently, but npm has been updated too..  so I don't know what is causing it.  but even if its npm, we STILL need to DO something to work around the destructive effects.

Jbebco3, 2019-09-20:
After I ran "git checkout modules/node_modules/node_helper/index.js" , I received an error message of:
"error: pathspec 'modules/node_modules/node_helper/index.js' did not match any file(s) known to git"
?? how badly does this affect my mirror's functionality, and what actions should I take to resolve this?

sdetweil, 2019-09-20:
this worked for me
from the MagicMirror folder
```
 git checkout modules/node_modules/node_helper/index.js
```

sdetweil, 2019-11-06:
how can I submit a pr, which updates the package.json, and creates a new separate repo under your userid?

sdetweil, 2019-11-14:
can i get you @MichMich to create a github repo with just the node_helper module.. then i can submit a pr change to the package.json to depend on  it.  see my comment on Sept 2 above

but the module has to BE someplace..

sdetweil, 2019-12-09:
can i get you @MichMich to create a github repo with just the node_helper module.. then i can submit a pr change to the package.json to depend on it. see my comment on Sept 2 above

sdetweil, 2019-12-09:
added my own repo

sdetweil, 2019-12-10:
pr #1827 submitted

sdetweil, 2019-12-29:
This should be closed. Fix in 2.10

closed 2019-12-29 by MichMich

johnnybfireman, 2021-12-27:
I am still having this issue.  Any help is greatly appreciated.
If I don't add any modules it works beautifully!

sdetweil, 2021-12-27:
@johnnybfireman  please don't post in multiple issues.. one will do..  I replied in the other

johnnybfireman, 2021-12-27:
Sorry my bad! Thank you for replying