MagicMirrorOrg/MagicMirror #1760: NPM Start command not working

24lbragg, 2019-08-31:
When I put in the command npm start to start magicmirror, i get this:

npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

> magicmirror@2.8.0 start /home/pi/MagicMirror
> sh run-start.sh

run-start.sh: 4: run-start.sh: electron: not found
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! magicmirror@2.8.0 start: `sh run-start.sh`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the magicmirror@2.8.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-08-31T15_34_25_041Z-debug.log

keep in mind that I am a complete noob and this is my first project.
Thanks, any help would be greatly appreciated.

sdetweil, 2019-09-01:
show us the output of npm -v

also try 
```
sudo npm i -g npm
```
then check version again

if it changed, then go to MagicMirror folder and do 
```
npm install
```
again

24lbragg, 2019-09-01:
npm -v output is 5.8.0. After the sudo npm i -g npm the version didn't change. 
After putting in the sudo, this is what i got:

npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

Usage: npm <command>

where <command> is one of:
    access, adduser, bin, bugs, c, cache, ci, completion,
    config, ddp, dedupe, deprecate, dist-tag, docs, doctor,
    edit, explore, get, help, help-search, i, init, install,
    install-test, it, link, list, ln, login, logout, ls,
    outdated, owner, pack, ping, prefix, profile, prune,
    publish, rb, rebuild, repo, restart, root, run, run-script,
    s, se, search, set, shrinkwrap, star, stars, start, stop, t,
    team, test, token, tst, un, uninstall, unpublish, unstar,
    up, update, v, version, view, whoami

npm <command> -h     quick help on <command>
npm -l           display full usage info
npm help <term>  search for help on <term>
npm help npm     involved overview

Specify configs in the ini-formatted file:
    /root/.npmrc
or on the command line via: npm <command> --key value
Config info can be viewed via: npm help config

npm@5.8.0 /usr/share/npm
\

Vicbouret, 2019-09-03:
You should try updating your Node version. I'm running 12.8.1 and it's working fine...

`curl -sL https://deb.nodesource.com/setup_12.x | bash -`
`sudo apt-get install nodejs`

Devan0369, 2019-09-03:
Hi! Everyone, its sound like I may have a similar problems however I am using Windows with Node js installed

Here is the logs
53 silly lifecycle magicmirror@2.8.0~install: Args: [ '/d /s /c', 'cd vendor && npm install' ]
54 timing audit submit Completed in 805ms
55 http fetch POST 200 https://registry.npmjs.org/-/npm/v1/security/audits/quick 806ms
56 timing audit body Completed in 5ms
57 silly lifecycle magicmirror@2.8.0~install: Returned: code: 0  signal: null
58 silly postinstall magicmirror@2.8.0
59 info lifecycle magicmirror@2.8.0~postinstall: magicmirror@2.8.0
60 verbose lifecycle magicmirror@2.8.0~postinstall: unsafe-perm in lifecycle true
61 verbose lifecycle magicmirror@2.8.0~postinstall: PATH: C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin;C:\Users\Devan Nair\Documents\Projects\Smart Mirror\MagicMirror\node_modules\.bin;C:\Users\Devan Nair\AppData\Roaming\npm;C:\Program Files\nodejs\;C:\Python36\Scripts\;C:\Python36\;C:\Program Files\Python36\Scripts\;C:\Program Files\Python36\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\nodejs\;C:\Users\Devan Nair\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files\JetBrains\PyCharm Community Edition 2019.1.2\bin;C:\Users\Devan Nair\AppData\Roaming\npm
62 verbose lifecycle magicmirror@2.8.0~postinstall: CWD: C:\Users\Devan Nair\Documents\Projects\Smart Mirror\MagicMirror
63 silly lifecycle magicmirror@2.8.0~postinstall: Args: [ '/d /s /c',
63 silly lifecycle   'sh installers/postinstall/postinstall.sh && npm run install-fonts' ]
64 silly lifecycle magicmirror@2.8.0~postinstall: Returned: code: 1  signal: null
65 info lifecycle magicmirror@2.8.0~postinstall: Failed to exec postinstall script
66 verbose stack Error: magicmirror@2.8.0 postinstall: `sh installers/postinstall/postinstall.sh && npm run install-fonts`
66 verbose stack Exit status 1
66 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:301:16)
66 verbose stack     at EventEmitter.emit (events.js:198:13)
66 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
66 verbose stack     at ChildProcess.emit (events.js:198:13)
66 verbose stack     at maybeClose (internal/child_process.js:982:16)
66 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
67 verbose pkgid magicmirror@2.8.0
68 verbose cwd C:\Users\Devan Nair\Documents\Projects\Smart Mirror\MagicMirror
69 verbose Windows_NT 6.3.9600
70 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install"
71 verbose node v10.16.3
72 verbose npm  v6.9.0
73 error code ELIFECYCLE
74 error errno 1
75 error magicmirror@2.8.0 postinstall: `sh installers/postinstall/postinstall.sh && npm run install-fonts`
75 error Exit status 1
76 error Failed at the magicmirror@2.8.0 postinstall script.
76 error This is probably not a problem with npm. There is likely additional logging output above.
77 verbose exit [ 1, true ]

can anyone please help

Almah2288, 2019-09-04:
The Solutions was to Uninstall Node.JS completely 

Install the most up to date Version.
if running Git
then open git 
-> $ npm -v
6.10.2

should show you build
continue 
to add this code

$ npm install -g @angular/cli

this will help open up that npm in Visual Code.

hopefully this work

lavolp3, 2019-09-04:
> run-start.sh: 4: run-start.sh: electron: not found
> npm ERR! file sh
> npm ERR! code ELIFECYCLE
> npm ERR! errno ENOENT
> npm ERR! syscall spawn
> npm ERR! magicmirror@2.8.0 start: `sh run-start.sh`
> npm ERR! [...]

Are you sure you are in the MagicmIrror/ directory when running npm start.
A safer way (on a standard raspberry pi install) is to try:
`cd ~/MagicMirror && npm start`

MichMich, 2019-10-04:
Should be solved in latest release. feel free to reopen.

closed 2019-10-04 by MichMich

24lbragg, 2019-10-12:
I tried some of your methods, and now it is showing this, I have no clue what to do.

pi@raspberrypi:~/MagicMirror $ npm start
/usr/lib/nodejs/aproba/index.js:1
(function (exports, require, module, __filename, __dirname) { ���������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������

SyntaxError: Invalid or unexpected token
    at new Script (vm.js:79:7)
    at createScript (vm.js:251:10)
    at Object.runInThisContext (vm.js:303:10)
    at Module._compile (internal/modules/cjs/loader.js:657:28)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)

Again, anything will help

sdetweil, 2019-10-12:
npm is downlevel 

do 

```
sudo npm i -g npm 
```
then try again

24lbragg, 2019-10-12:
showed the same thing when i put it in and also when i tried again.

kennymanman, 2020-03-02:
am having issues with my npm start. please i need help

C:\Users\hp\sample> npm start
npm ERR! missing script: start

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\hp\AppData\Roaming\npm-cache\_logs\2020-03-02T20_53_25_568Z-debug.log

C:\Users\hp\sample>

sdetweil, 2020-03-02:
C:\Users\hp\sample> npm start

u need to be in the folder created by the install, typically called MagicMirror
the folder will contain the package.json file for the MagicMirror app, and the 'start' script is defined there