MagicMirrorOrg/MagicMirror #1764: Error: Cannot find module 'node_helper'

tanloi123, 2019-09-09:
> magicmirror@2.8.0 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.8.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'node_helper'
Loading module helpers ...
No helper found for module: alert.
App threw an error during load
Error: Cannot find module 'node_helper'
    at Module._resolveFilename (internal/modules/cjs/loader.js:602:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (internal/modules/cjs/loader.js:528:25)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:6:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:94:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
Whoops! There was an uncaught exception...
{ Error: Cannot find module 'node_helper'
    at Module._resolveFilename (internal/modules/cjs/loader.js:602:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (internal/modules/cjs/loader.js:528:25)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:6:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:94:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32) code: 'MODULE_NOT_FOUND' }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.
Shutting down server...

tanloi123, 2019-09-09:
pls help me!!

sdetweil, 2019-09-09:
do this to get the file back

```
cd ~/MagicMirror
git checkout modules/node_modules/node_helper/index.js
```

thundergreen, 2019-09-20:
Problem is we run this as hassio addon. How would we make this then @sdetweil ?

sdetweil, 2019-09-20:
you could download the file manually and somehow put it in the right location.  ssh into the hassio machine maybe?

thundergreen, 2019-09-20:
@sdetweil  I found a solution by just adding 

```
"cmd": "cd $MIRROR_APP_PATH/ && git checkout modules/node_modules/node_helper/index.js"
```
to my docker addon. And it works fine

closed 2019-10-04 by MichMich