MagicMirrorOrg/MagicMirror #1769: In the process of installation, I always encounter errors when installing, and many people also encounter such a clear departure.

Dqaq, 2019-09-14:
npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EAI_AGAIN: request to https://registry.npmjs.org/grunt failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443
npm WARN registry Using stale data from https://registry.npmjs.org/ due to a request error during revalidation.
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm ERR! Unexpected end of JSON input while parsing near '...{"node":">=0.6.0"}},"'

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-14T09_34_31_079Z-debug.log

Joshua2504, 2019-09-19:
Could you please post the content of file

`/home/pi/.npm/_logs/2019-09-14T09_34_31_079Z-debug.log`

sdetweil, 2019-09-20:
Dqaq, did you use the installer script, or do it manually? (git clone and then npm install)?

can u show us the output of node -v and npm -v

the error is usually a downlevel npm 

try the updated installer, see here
```
https://forum.magicmirror.builders/topic/10171/anyone-want-to-try-updated-installer
```

sdetweil, 2019-11-14:
@MichMich  you should remove the PR Welcome tag, as u have merged a PR

sdetweil, 2019-12-29:
This should be closed, as the fixes are in 2.10

closed 2019-12-29 by MichMich