MagicMirrorOrg/MagicMirror #1771: Install and run the app with: npm install && npm start

Darkmx-6, 2019-09-26:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi Zero w running DietPi v6.25.3

**Node Version**: ?

**MagicMirror Version**:  v2

**Description**: During install I get a  Error: Cannot find module 'semver' when I do step 4 on the manual Install : npm install && npm start  

This happened on both the manual and bash command

**Steps to Reproduce**: step 4: Install and run the app with: npm install && npm start

**Expected Results**: I just want magic mirror to work

**Actual Results**: root@DietPi:~# bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"

$$\      $$\                     $$\           $$\      $$\ $$\                                          $$$$$$\
$$$\    $$$ |                    \__|          $$$\    $$$ |\__|                                        $$  __$$\
$$$$\  $$$$ | $$$$$$\   $$$$$$\  $$\  $$$$$$$\ $$$$\  $$$$ |$$\  $$$$$$\   $$$$$$\   $$$$$$\   $$$$$$\  \__/  $$ |
$$\$$\$$ $$ | \____$$\ $$  __$$\ $$ |$$  _____|$$\$$\$$ $$ |$$ |$$  __$$\ $$  __$$\ $$  __$$\ $$  __$$\  $$$$$$  |
$$ \$$$  $$ | $$$$$$$ |$$ /  $$ |$$ |$$ /      $$ \$$$  $$ |$$ |$$ |  \__|$$ |  \__|$$ /  $$ |$$ |  \__|$$  ____/
$$ |\$  /$$ |$$  __$$ |$$ |  $$ |$$ |$$ |      $$ |\$  /$$ |$$ |$$ |      $$ |      $$ |  $$ |$$ |      $$ |
$$ | \_/ $$ |\$$$$$$$ |\$$$$$$$ |$$ |\$$$$$$$\ $$ | \_/ $$ |$$ |$$ |      $$ |      \$$$$$$  |$$ |      $$$$$$$$\
\__|     \__| \_______| \____$$ |\__| \_______|\__|     \__|\__|\__|      \__|       \______/ \__|      \________|
                       $$\   $$ |
                       \$$$$$$  |
                        \______/

this appears not to be a Raspberry Pi 2 or 3, do you want to continue installtion (y/N)?y
Updating packages ...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease                
Hit:2 https://archive.raspberrypi.org/debian buster InRelease                  
Reading package lists... Done                            
Installing helper tools ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.6).
curl is already the newest version (7.64.0-4).
git is already the newest version (1:2.20.1-2).
unzip is already the newest version (6.0-23+deb10u1).
wget is already the newest version (1.20.1-1.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Check current Node installation ...
Node currently installed. Checking version number.
Minimum Node version: v5.1.0
Installed Node version: v8.11.3
No Node.js upgrade necessary.
Check current NPM installation ...
NPM currently installed. Checking version number.
module.js:549
    throw err;
    ^

Error: Cannot find module 'semver'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/share/npm/lib/utils/unsupported.js:2:14)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
[... 108 lines cut ...]
 kernel:[ 4500.914127] df00: c09e3028 2c58bbb1 d58bfaa0 00000000 00000000 d5958aa0 d4f0bd54 00000002

Message from syslogd@raspberrypi at Sep 24 18:13:12 ...
 kernel:[ 4500.921870] df20: 00000000 d5958aa0 d595df7c d595df38 c01a8508 c01087d4 ffffffff 7fffffff

Message from syslogd@raspberrypi at Sep 24 18:13:12 ...
 kernel:[ 4500.929337] df40: 00000000 0000009a ffffffff 7fffffff d595df94 00000000 00000000 beb49ad0

Message from syslogd@raspberrypi at Sep 24 18:13:12 ...
 kernel:[ 4500.937129] df60: 00000155 c00091a4 d595c000 00000000 d595dfa4 d595df80 c01a8594 c01a83b0

Message from syslogd@raspberrypi at Sep 24 18:13:12 ...
 kernel:[ 4500.944709] df80: 00000000 00000000 00000002 d595df98 00000000 00000000 00000000 d595dfa8

Message from syslogd@raspberrypi at Sep 24 18:13:12 ...
 kernel:[ 4500.952390] dfa0: c0009000 c01a8578 00000000 00000000 0000000b 00000002 00000000 00000000

Message from syslogd@raspberrypi at Sep 24 18:13:12 ...
 kernel:[ 4500.960192] dfc0: 00000000 00000000 beb49ad0 00000155 0159af66 00de4280 ffffffff 00de4218

Message from syslogd@raspberrypi at Sep 24 18:13:12 ...
 kernel:[ 4500.967783] dfe0: 0000006c beb496c0 00013de8 b6ef4d0c 60000010 0000000b 00000000 00000000

Message from syslogd@raspberrypi at Sep 24 18:13:12 ...
 kernel:[ 4501.075602] Code: e1a02000 e1a01005 e5980000 ebfd77db (e1c821d0)

MichMich, 2019-10-04:
Thanks for opening an issue!

To allow us to help you it's extremely important that we have all the details about your setup. During the creation of this issue, important details were asked to fill in. Because you did not provide (any) of these details we are unable to help you. 

If you are running into errors, please provide as many detailed error logs as possible in addition to the requested details.

If you are unaware how to retrieve all the requested information, feel free to ask for help on http://forum.magicmirror.builders.

This issue will be closed. Feel free to open a new issue with all the requested information.

closed 2019-10-04 by MichMich

Darkmx-6, 2019-10-04:
Is this a canned response? Because I did have lots of details in my post,
may be I didn't have all the details asked for but I had most.

Darkmx-6, 2019-10-04:
Anyhow the problem I was having was because of some shorted GPIO pins, killing all GPIOs and causing the kernel errors.

It now works and I'm having a problem auto starting videos but that is a problem for another thread.