MagicMirrorOrg/MagicMirror #1772: Undefined at location Weatherforecast

maxduinhoven, 2019-09-26:
Replace 210 for:  if (this.config.appendLocationNameToHeader && this.data.header !== undefined) {
solves the problem for me

https://github.com/MichMich/MagicMirror/blob/a31546b1ff8c7c10e12d7acbab063297d59c6b93/modules/default/weatherforecast/weatherforecast.js#L206-L215

[image]
[image]

MichMich, 2019-10-04:
Please send a PR!

closed 2019-10-04 by MichMich

WorldWideWebDev, 2020-12-18:
I know this is old but recently I had the same problem and it seemed in my case that i was missing a line named "header" in one of the sections , see below.
I added the line(( header: "Current Weather" ))and it showed what i wrote. you can add a header as you wish and it seems to work.

      }
                                ]
                        }
                },
                {
                        module: "compliments",
                        position: "lower_third"
                },
                {
                        module: "currentweather",
                        position: "top_right",
                        header: "Current Weather",     (// this section can be anything you wish and it removes  "UNDEFINED ") 
                        config: {
                                location: "Graz,Austria",
                                locationID: "", 
                                appid: "5f3624435d345s6fg2e32o120"
                        }
                },
                {
                        module: "weatherforecast",
                        position: "top_right",
                        header: "Weather Forecast",   (// this section can be anything you wish and it removes  "UNDEFINED ")
                        config: {
                                location: "Graz,Austria",
                                locationID: "", 
                                appid: "5f3624435e4tyu4b542e32o120"
                        }
                },
                {
                        module: "newsfeed",
                        position: "bottom_bar",
                        config: {
                                feeds: [
                                        {
                                                title: "Google, Duetsche",
                                                url: "https://news.google.com/news/rss/?ned=de&$
                                        },
                                        {
                                                title: "Sued Deutsche",

WorldWideWebDev, 2020-12-18:
I have experienced this problem also. I have three MM's and two experience this , obviously because the two are mirrored SD cards with identical operating systems. But i overcame the issue of it displaying "Undefined" by adding a line under the "position line" named " header" copied from some lines below. I added (header: "Current Weather",) which was missing. Note its missing and obviously not required as all three did not have this line in the "cuurentweather" section but one of them populates and the other two do not populate and therefore display the word "undefined" . So the question remains for me "Where is this line populated from?" 
Anyway by adding the line and writing what you wish, the word "undefined" goes away and what you have written is shown. Good until i break it for some other reason. :) Note - this line obviously is not supposed to be there and this is a workaround, until i work it out or start again. 

    module: "compliments",
                        position: "lower_third"
                },
                {
                        module: "currentweather",
                        position: "top_right",
                        header: "Current Weather Graz",    ----- (I added this line and now it displays what i wrote) -----
                        config: {
                                location: "Graz,Austria",
                                locationID: "", 
                                appid: "cf360985ae1d34567sg42e32b120"
                        }
                },
                {
                        module: "weatherforecast",
                        position: "top_right",
                        header: "Weather Forecast",
                        config: {
                                location: "Graz,Austria",
                                locationID: "", 
                                appid: "qf30987dfe1d6b25fe63456a120"
                        }
                },
                {
                        module: "newsfeed",
                        position: "bottom_bar",
                        config: {
                                feeds: [
                                        {
                                                title: "Google, Duetsche",
                                                url: "https://news.google.com/news/rss/?ned=de&$
                                        },
                                        {
                                                title: "Sued Deutsche",