MagicMirrorOrg/MagicMirror #1778: Udating problem

That-Guy-Jack, 2019-10-03:
Platform: rasparry pi 3 b in pi top (laptop edition) raspian (think its most recent lees than 2 months old install )

**Node Version**: dont know how to find 
**MagicMirror Version**: Magic mirror 2 

**Description**: cant update thorugh SSH and on the pi directly getting error 

**Steps to Reproduce**: just try updating 

**Expected Results**: it to update

**Actual Results**: this error pi@raspberrypi:~/MagicMirror $ git pull && npm install
Updating a31546b..500147e
error: Your local changes to the following files would be overwritten by merge:
        package-lock.json
Please commit your changes or stash them before you merge.
Aborting
pi@raspberrypi:~/MagicMirror $

**Configuration**: basicly stock 

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

sdetweil, 2019-10-03:
try my new upgrade script,  see here in the form for instructions

https://forum.magicmirror.builders/topic/10859/new-update-upgrade-script-ready-for-testing

That-Guy-Jack, 2019-10-03:
> 
> 
> try my new upgrade script, see here in the form for instructions
> 
> https://forum.magicmirror.builders/topic/10859/new-update-upgrade-script-ready-for-testing

ill try that in a sec thanks

That-Guy-Jack, 2019-10-03:
> 
> 
> try my new upgrade script, see here in the form for instructions
> 
> https://forum.magicmirror.builders/topic/10859/new-update-upgrade-script-ready-for-testing

seemed to have fixed it by doingthant and removing the lock file as thier was nothing in it soo that has fixed it thanks for your help

sdetweil, 2019-10-03:
cool.. can u close this issue now?

closed 2019-10-04 by That-Guy-Jack