MagicMirrorOrg/MagicMirror #1797: After installing a module only ablack screen is displayed

kk69bs, 2019-10-28:
On latest Rasbian version the new MagicMirror installation worked fine until I added any external module. The screen shows only a black screen. Here is the terminal response:

pi@KPI5:~/MagicMirror $ DISPLAY=:0 npm start

> magicmirror@2.9.0 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.9.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'node_helper'
Loading module helpers ...
No helper found for module: alert.
App threw an error during load
Error: Cannot find module 'node_helper'
    at Module._resolveFilename (internal/modules/cjs/loader.js:602:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (internal/modules/cjs/loader.js:528:25)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:6:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:100:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
Whoops! There was an uncaught exception...
{ Error: Cannot find module 'node_helper'
    at Module._resolveFilename (internal/modules/cjs/loader.js:602:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (internal/modules/cjs/loader.js:528:25)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:6:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:100:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32) code: 'MODULE_NOT_FOUND' }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.
^C

Thanks for any helping hint!

Claudiop193w, 2019-10-29:
I also have this problem .. can I get a link of version 2.8 ??  I know how to get back from 2.9 to the previous version

sdetweil, 2019-10-29:
sounds like you did an npm install in a module folder without a package.json file..

do this to resolve
```
cd ~/MagicMirror
git checkout -- modules/node_modules/node_helper/index.js
```

mishu28nmv, 2019-11-03:
If you disable module ALERT, I think it will be fine. I'm just a beginner but maybe this will do the trick!

MichMich, 2019-11-05:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-11-05 by MichMich

Claudiop193w, 2019-11-06:
continuo ad avere problemi con la 2.9......
 pm2 non parte si blocca continuamente in precedenza non ho mai avuto nessun problema

johnnybfireman, 2021-12-27:
I see these posts are from 2019 and I am still having the problem.
When I run 
`
cd ~/MagicMirror
git checkout -- modules/node_modules/node_helper/index.js
`
I get the error:
`
````
git checkout modules/node_modules/node_helper/index.js
error: pathspec "modules/node_modules/node_helper/index.js'\ did not match any file(s) known to git```

If I run npm start dev:
```
    at Module._resolveFilename (internal/modules/cjs/loader.js:887:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:257:1128)
    at Function.Module._resolveFilename (/home/john/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (internal/modules/cjs/loader.js:732:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Module.require (internal/modules/cjs/loader.js:959:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/home/john/MagicMirror/modules/MMM-bitcoin/node_helper.js:2:15)
    at Module._compile (internal/modules/cjs/loader.js:1078:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1108:10)
[27.12.2021 11:46.33.394] [ERROR] Whoops! There was an uncaught exception...
[27.12.2021 11:46.33.404] [ERROR] Error: Cannot find module 'request'
Require stack:
- /home/john/MagicMirror/modules/MMM-bitcoin/node_helper.js
- /home/john/MagicMirror/js/app.js
- /home/john/MagicMirror/js/electron.js
- /home/john/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:887:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:257:1128)
    at Function.Module._resolveFilename (/home/john/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (internal/modules/cjs/loader.js:732:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Module.require (internal/modules/cjs/loader.js:959:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/home/john/MagicMirror/modules/MMM-bitcoin/node_helper.js:2:15)
    at Module._compile (internal/modules/cjs/loader.js:1078:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1108:10) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/john/MagicMirror/modules/MMM-bitcoin/node_helper.js',
    '/home/john/MagicMirror/js/app.js',
    '/home/john/MagicMirror/js/electron.js',
    '/home/john/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[27.12.2021 11:46.33.409] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[27.12.2021 11:46.33.410] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[27.12.2021 11:46.33.726] [LOG]   Launching application.````

Any suggestions are greatly appreciated.

sdetweil, 2021-12-27:
see https://forum.magicmirror.builders/topic/15778/fix-for-black-screen-in-2-16-and-later