MagicMirrorOrg/MagicMirror #1800: electron downlevel on Raspian Buster

sdetweil, 2019-10-31:
**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).
Raspian Buster 

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).
2.9.0
**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.
electron cannot start 
```
 magicmirror@2.9.0 start /home/pi/MagicMirror
> sh run-start.sh

/home/pi/MagicMirror/node_modules/electron/dist/electron: /lib/arm-linux-gnueabihf/libdbus-1.so.3: no version information available (required by /home/pi/MagicMirror/node_modules/electron/dist/electron)
/home/pi/MagicMirror/node_modules/electron/dist/electron: /usr/lib/arm-linux-gnueabihf/libnss3.so: version `NSS_3.22' not found (required by /home/pi/MagicMirror/node_modules/electron/dist/electron)
```
**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

Electron project says upgrade electron is only fix.

MichMich, 2019-11-05:
So, what version do we need to bump to? Feel free to send a PR.

sdetweil, 2019-11-06:
working on that. 6.0.12 seems good..

MichMich, 2020-01-01:
As discussed, the version change of Electron was reverted because of Travis issues. Electron definitely needs to be updated, so we need to find a way to make it work with Travis.

@roramirez Any idea?

MichMich, 2020-01-04:
For anyone running Buster, if you want to switch to a working Electron version, use the following command:
```
npm install electron@6.0.12
```

Please report if this solves your issue. Thanks!

khassel, 2020-01-06:
concerning the electron issue, you are talking about the failed tests [here](https://travis-ci.com/MichMich/MagicMirror/builds/142809334)?

I did tests with electron 6.0.12, 6.1.7 and 7.1.7. Tests with version 6 are failing, but with 7.1.7 everything is fine.

MichMich, 2020-01-06:
Does 7.1.7 work on a Raspberry? Haven't tried it yet. If so, feel free to send a PR, and let's check if Travis works.

sdetweil, 2020-01-06:
pushed wrong button

sdetweil, 2020-01-06:
no.. 7.1.7 will not install on PI4 buster

```
> electron@7.1.7 postinstall /home/pi/MagicMirror/node_modules/electron
> node install.js

(node:7484) UnhandledPromiseRejectionWarning: HTTPError: Response code 404 (Not Found)
    at EventEmitter.emitter.on.response (/home/pi/MagicMirror/node_modules/got/source/as-stream.js:35:24)
```

khassel, 2020-01-06:
Sam is correct, pi doesn't  run with 7.

Is 5.0.13 an option? Tests passes and pi is working.

sdetweil, 2020-01-06:
6.0.12 looks good, but Travis env needs fixed

khassel, 2020-01-06:
I did the tests without Travis, they fail with 6.0.12 or 6.1.7. Same tests are o.k. with 7.1.7 and 5.0.13.

So I'm no electron expert, no idea whats going wrong with 6. So if noone has an idea to get the tests running with 6 we have the choice between
* use 6 and disable the tests
* use 5

sdetweil, 2020-01-06:
the tests work locally if u export DISPLAY=:0

roramirez, 2020-01-07:
What version of Electron we'll expect to set for the project? I'll try to get a free window time this week to working on in this and find a something can work.

sdetweil, 2020-01-07:
i think we a have to go to 6, thats the 1st version that supports the current node 10, and  electron 7 doesn't work.

I tested with 6.0.12, which seemed stable and reliable for all the modules I tested with. 

there were breaking changes to go to 6, but the MM core has already adjusted. 

there is a bug somewhere in the vendor tests, which launch MM using config of port 8080, but actually use 9515. i posted info in some issue.  change the urls in testcase to use port 9515 and they work

khassel, 2020-01-07:
can confirm that (with updated spectron) the tests passes with electron v6, may we need adjust the 10 sec. in the weather tests, they fail sometimes because of timeout.

I tested with v6.1.7, this is the latest v6, see Version History [here](https://www.npmjs.com/package/electron)

I'm not shure if 6.0.12 is still getting updates:
````
6.1.7   22 days ago
6.0.12  3 months ago
````

sdetweil, 2020-01-07:
>I'm not shure if 6.0.12 is still getting updates:

we were on 3.0.13, LONG LONG since any updates.. i suspect we will update maybe once a year..

roramirez, 2020-01-08:
> there is a bug somewhere in the vendor tests, which launch MM using config of port 8080, but actually use 9515. i posted info in some issue. change the urls in testcase to use port 9515 and they work

Do you have the # for this issue?

sdetweil, 2020-01-08:
@roramirez https://github.com/MichMich/MagicMirror/issues/1842

khassel, 2020-01-16:
Did some investigation in the weather tests (with electron v6.x). This seems to be a bug in v6.x, they fail whatever is coded as timeout.

So I decided to investigate in the electron v7 problem, where `npm install` fails on a raspberry. The solution here is to use `npm install --arch=armv7l`.

Tested this fix on a raspberry, and with v7 the weather tests are o.k.

So if someone can confirm these results, this could be the solution for the electron problem.

Here some logs:
* [log of debian build](https://gitlab.com/khassel/magicmirror/-/jobs/406151376)
* [log of rpi build](https://gitlab.com/khassel/magicmirror/-/jobs/406151377)
* [log of tests based on debian build](https://gitlab.com/khassel/magicmirror/-/jobs/406151378)

khassel, 2020-01-17:
Did some updates of the dependencies in the `package.json`. The optimistic approach, setting all to `"latest"` didn't work (expected ...), but nearly. Only `"jsdom"` must stay on 11.x.

So a new `package.json` could look like
````
{
  "name": "magicmirror",
  "version": "2.10.0",
  "description": "The open source modular smart mirror platform.",
  "main": "js/electron.js",
  "scripts": {
    "start": "./run-start.sh",
    "install": "./install.sh",
    "install-fonts": "cd fonts && npm install",
    "postinstall": "sh untrack-css.sh && sh installers/postinstall/postinstall.sh && npm run install-fonts",
    "test": "NODE_ENV=test ./node_modules/mocha/bin/mocha tests --recursive",
    "test:unit": "NODE_ENV=test ./node_modules/mocha/bin/mocha tests/unit --recursive",
    "test:e2e": "NODE_ENV=test ./node_modules/mocha/bin/mocha tests/e2e --recursive",
    "config:check": "node tests/configs/check_config.js",
    "lint": "grunt"
  },
  "repository": {
    "type": "git",
    "url": "git+https://github.com/MichMich/MagicMirror.git"
  },
  "keywords": [
    "magic mirror",
    "smart mirror",
    "mirror UI",
    "modular"
  ],
  "author": "Michael Teeuw",
  "contributors": [
    "https://github.com/MichMich/MagicMirror/graphs/contributors"
  ],
  "license": "MIT",
  "bugs": {
    "url": "https://github.com/MichMich/MagicMirror/issues"
  },
  "homepage": "https://magicmirror.builders",
  "devDependencies": {
    "chai": "latest",
    "chai-as-promised": "latest",
    "current-week-number": "latest",
    "danger": "latest",
    "grunt": "latest",
    "grunt-eslint": "latest",
    "grunt-jsonlint": "latest",
    "grunt-markdownlint": "latest",
    "grunt-stylelint": "latest",
    "grunt-yamllint": "latest",
    "http-auth": "latest",
    "jsdom": "^11.12.0",
    "jshint": "latest",
    "mocha": "latest",
    "mocha-each": "latest",
    "mocha-logger": "latest",
    "spectron": "latest",
    "stylelint": "latest",
    "stylelint-config-standard": "latest",
    "time-grunt": "latest"
  },
  "optionalDependencies": {
    "electron": "latest"
  },
  "dependencies": {
    "colors": "latest",
    "console-stamp": "latest",
    "express": "latest",
    "express-ipfilter": "latest",
    "feedme": "latest",
    "helmet": "latest",
    "iconv-lite": "latest",
    "lodash": "latest",
    "module-alias": "latest",
    "moment": "latest",
    "request": "latest",
    "rrule": "latest",
    "rrule-alt": "latest",
    "simple-git": "latest",
    "socket.io": "latest",
    "valid-url": "latest"
  },
  "_moduleAliases": {
    "node_helper": "js/node_helper.js"
  }
}
````

where `install.sh` is

````
#!/bin/bash

cd vendor

if [[ $(arch) =~ ^arm ]]; then
  npm install --arch=armv7l
else
  npm install
fi;
````

So this could be the content of a pull request solving the electron problem and updating the dependencies.

sdetweil, 2020-01-18:
what will we have to do each quarter to retest that latest is ok?

khassel, 2020-01-18:
same as now. You have already 1/3 "latest" in the current `package.json`.

Its a discussion how getting updates without breaking the app. May one approach is to work with a "latest-package.json" on the `develop` branch an nail the versions with a release to real version numbers, so the `package.json` on `master` has only fix version numbers and is only updated with new releases.

MichMich, 2020-01-18:
I'm not a fan of using `latest` in this case, since there is a big chance a new version of a dependency doesn't work in a electron/raspberry environment. I prefer to set the versions that work on the supported platform and only upgrade if necessary.

MichMich, 2020-01-18:
Also note that the install.sh script will be removed in the near future.

khassel, 2020-01-18:
so how to continue with electron? Update to 7.x with spectron 9.x? This works only with install.sh or a similar construction.

MichMich, 2020-01-18:
is `npm install --arch=armv7l` necessary on the raspberry or on travis?

khassel, 2020-01-18:
this is necessary on the raspberry only.

MichMich, 2020-01-18:
Bummer, otherwise it could be part of the travis script. Hopefully there will be a more elegant solution available before 2019/4/1.

khassel, 2020-01-18:
One solution is to put this extra line into `package.json`:

````
    "preinstall": "if arch | grep -Eq '^armv.*' ; then npm install electron --arch=armv7l; fi",
````

This installs electron before the "main" install, only for electron the arch-param is needed.

sdetweil, 2020-01-19:
doesn't this run the risk of having some modules at arvm7l (required by electron) and others not?
I know we have tons of problems with grpc, and epoll, and snowboy, which are pre-built by architecture and new installs/upgrades mess them up cause they don't run inside electron  

I tried the --arch=armv7l to get around the pi 0 (armv6l) lack of running versions, but that really didn't work out well.

sdetweil, 2020-01-19:
@MichMich  >before 2019/4/1.

u meant 2020/4/1!

khassel, 2020-01-19:
> doesn't this run the risk of having some modules at arvm7l (required by electron) and others not?

no, building electron v7 with the arch param is a workaround for [this problem](https://github.com/electron/electron/issues/20723#issuecomment-556915720).

> I tried the --arch=armv7l to get around the pi 0 (armv6l) lack of running versions, but that really didn't work out well.

the future of pi 0 is already answered by @MichMich [here](https://github.com/MichMich/MagicMirror/issues/1860#issuecomment-570785138).

Another "solution" instead of using preinstall in package.json could be only updating the install docs with
* if you want to install on a raspi 0 use ...
* if you want to install on a raspi > 0 use `npm install --arch=armv7l`
* in all other cases use `npm install`

sdetweil, 2020-01-19:
I already have that in the automated script. Just have to invert the test for armv6l

roramirez, 2020-01-20:
I pushed the Pull Request #1892 using a new version of Electron and the testing for e2e are passing except the weather module. I will take a look after this PR if is merged because is not a related issue here.

I think to move the 7 version of Electron is too rush. Also I'm not a big fan of the newest versions

khassel, 2020-01-20:
as said before, weather tests are not running with electron v6, but e.g. with v7.
The function `waitForExist` is only used in the weather tests, maybe thats related to the problem in v6.

roramirez, 2020-01-21:
> as said before, weather tests are not running with electron v6, but e.g. with v7.
> The function `waitForExist` is only used in the weather tests, maybe thats related to the problem in v6.

I think we'll should keep in the 6 version and fix the weather test, after that may be will move to 7 version of Electron.

How  I see it with the 6 version there are doesn't troubles with Raspberry model, but really really not sure

sdetweil, 2020-01-21:
@roramirez   the vendor tests also did not correctly execute as per #1842 

```
Vendors
    Get list vendors
      ✓ should return 200 HTTP code for vendor "moment.js"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "moment-timezone.js"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "weather-icons.css"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "weather-icons-wind.css"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "font-awesome.css"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "nunjucks.js"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "suncalc.js"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"moment.js"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"moment-timezone.js"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"weather-icons.css"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"weather-icons-wind.css"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"font-awesome.css"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"nunjucks.js"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"suncalc.js"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
```

sdetweil, 2020-03-21:
waiting on release 2.11 for proposed fix

sdetweil, 2020-04-05:
2.11 ships 6.1.7

closed 2020-04-05 by sdetweil