MagicMirrorOrg/MagicMirror #1802: Graphical user interface

halfajobj, 2019-11-04:
Good Afternoon

I was woundering if there are any plans to to create  GUI for magic mirror so its easier to set up like the Dac borad with option for using code to modify if you so wish

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

MichMich, 2019-11-05:
Not currently planned. I think there are some 3rd party initiatives that hook into MagicMirror². Please check the forum: http://forum.magicmirror.builders

closed 2019-11-05 by MichMich

brettjbyers, 2020-11-26:
Anyone know of a place I can hire someone to help me with some basic module setup?  I’m terrible at this stuff, and going though YouTube videos hasn’t helped much.  I have gone as far as getting my API key at openweathermap but modifying the files is beyond me.

sdetweil, 2020-11-26:
see this , its easy... 

https://forum.magicmirror.builders/topic/4231/how-to-add-modules-for-absolute-beginners

JibGeonzon, 2023-12-07:
> Anyone know of a place I can hire someone to help me with some basic module setup? I’m terrible at this stuff, and going though YouTube videos hasn’t helped much. I have gone as far as getting my API key at openweathermap but modifying the files is beyond me.

I can help you make one.