MagicMirrorOrg/MagicMirror #1803: Recurring Google Calendar events for times when UTC date differ from local date appear at wrong time in MM

nerdoug, 2019-11-06:
**Platform**: Raspi 2, current versions from automatic build of MM as of 2019-10-21.
pi@raspberrypi:~/MagicMirror$ uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

**Node Version**: pi@raspberrypi:~/MagicMirror$ node -v
v8.11.1

**MagicMirror Version**: pi@raspberrypi:~/MagicMirror$ head -3 package.json |tail -2
  "name": "magicmirror",
  "version": "2.9.0",

**Description**:  The problem concerns Google Calendar events which are recurring, and have a start time in the period at the end of the day local time when the UTC date is different than local time. In this case, the MM calendar display will show the event either 24 hours early, or perhaps not at all in the limited list of events that are displayed.

**Steps to Reproduce**: 
1) determine the period for your timezone when the UTC date is different than the local date ("PWDD" to save typing). 

If you live West of Greenwich, England, the PWDD starts at midnight plus the offset for your timezone (which may be negative.) I live near Toronto Canada, and we've just gone back from Daylight Saving time (offset -4) to Standard time (offset -5) so our PWDD now starts at 7PM local time. The PWDD ends at midnight local time for this case.

If you live East of Greenwich England, your PWDD starts at midnight local time, and ends at Midnight plus your UTC offset.

2) define 4 test events, each 30 minutes long in your Google calendar, with a one week recurrence:
      A) 30 minutes before the start of your PWDD
      B) 30 minutes after the start of your PWDD
      C) 30 minutes before the end of your PWDD
      D) 30 minutes after the end of your PWDD

3) Give MM calendar enough time to update, and check to see how these events appear in the MM calendar. In my testing the events within the PWDD, B) and C) above, are either shown 24 hours early, or not at all.

4) In Google Calendar, change all 4 events to be non-recurring.

5) Give MM calendar enough time to update, and check how these events appear in the MM calendar. In my testing, they're always all correctly displayed.

**Expected Results**: 
Event start times in Google Calendar and MM Calendar should agree

**Actual Results**: 
Event start times are incorrect, 24 hours too early, or the event does not appear at all in the limited amount of calendar displayed.

**Configuration**: Will add this a bit later.

**Additional Notes**: 
-more testing results in issue 1105, after Oct 21.
-glad to provide more files, iCal curls, or do testing

tisnatch, 2019-11-06:
Can confirm this bug is affecting me too.

nerdoug, 2019-11-06:
Here's part of the top of my config.js file
```
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
  */

var config = {
        address: "localhost", 
        port: 8080,
        ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], 
        language: "en",
        timeFormat: 24,
        units: "metric",

        modules: [
```

Here's the calendar section from my config.js
```
{
   module: "calendar",
   header: "Upcoming Stuff",
   position: "top_left",
   config: {
              colored: false,
              displaySymbol: false,
              // :wqcoloredSymbolOnly: true,
              timeFormat: "absolute",
              getRelative: 0,
              urgency: 0,
              dateFormat: "ddd MMM DD hh:mm a",
              showEnd: false,
              maximumEntries: 15,
              maxTitleLength: 30,
              wrapEvents: true,
              fade: false,
              broadcastEvents: false,
              animationSpeed: 0,
              fetchInterval: 120000,
              calendars: [
               {
                 // symbol: "calendar-check",
                 url: "webcal://www.calendarlabs.com/ical-calendar/ics/39/Canada_Holidays.ics"
               },
               {
                 // symbol: "calendar-check",
                 url: "https://calendar.google.com/calendar/ical/xxx%40group.calendar.google.com/public/basic.ics",
                 color: "#FF0000"
               }
                         ]
           }
},
```
Here's the iCal code for my test event at 19:30 local, 00:30 UTC, as a non-recurring event, that displayed properly. Note the timestamps in UTC.
```
BEGIN:VEVENT
DTSTART:20191107T003000Z
DTEND:20191107T010000Z
DTSTAMP:20191106T031814Z
UID:xxxxxx@google.com
CREATED:20191105T230644Z
DESCRIPTION:
LAST-MODIFIED:20191105T231141Z
LOCATION:
SEQUENCE:1
STATUS:CONFIRMED
SUMMARY:testB
TRANSP:OPAQUE
END:VEVENT
```
Here is the iCal code for the same event, change to recurring weekly, which displays incorrectly in MM. Note the timestamps relative to my time zone, America/Toronto, and the addition of the RRULE tag describing the recurrence. (I've seen other events that use this type of timestamp that seem to work fine in MM.)

```
BEGIN:VEVENT
DTSTART;TZID=America/Toronto:20191106T193000
DTEND;TZID=America/Toronto:20191106T200000
RRULE:FREQ=WEEKLY;BYDAY=WE
DTSTAMP:20191106T032446Z
UID:xxxxxxx@google.com
CREATED:20191105T230644Z
DESCRIPTION:
LAST-MODIFIED:20191106T032421Z
LOCATION:
SEQUENCE:2
STATUS:CONFIRMED
SUMMARY:testB
TRANSP:OPAQUE
END:VEVENT
```
Here's the beginning of the iCal curl, which includes the time zone definitions
```
BEGIN:VCALENDAR
PRODID:-//Google Inc//Google Calendar 70.9054//EN
VERSION:2.0
CALSCALE:GREGORIAN
METHOD:PUBLISH
X-WR-CALNAME:pc: canoe.eh@gmail.com
X-WR-TIMEZONE:America/Toronto
X-WR-CALDESC:google calendar\, hopefully
BEGIN:VTIMEZONE
TZID:America/Toronto
X-LIC-LOCATION:America/Toronto
BEGIN:DAYLIGHT
TZOFFSETFROM:-0500
TZOFFSETTO:-0400
TZNAME:EDT
DTSTART:19700308T020000
RRULE:FREQ=YEARLY;BYMONTH=3;BYDAY=2SU
END:DAYLIGHT
BEGIN:STANDARD
TZOFFSETFROM:-0400
TZOFFSETTO:-0500
TZNAME:EST
DTSTART:19701101T020000
RRULE:FREQ=YEARLY;BYMONTH=11;BYDAY=1SU
END:STANDARD
END:VTIMEZONE
```
Hope this is helpful.

tisnatch, 2019-11-06:
My apologies, actually, it's working fine for me. I can't confirm this issue. My recurring events work find in the core calendar module.  I'm having issues with recurring events in MMM-CalendarWeek. I had incorrectly assumed that module used some of the same back-end iCal plumbing as the core module. But it doesn't appear as such. I'm going to log an issue on MMM-CalendarWeek and reference this one. Maybe their code is based on this module so if there's a bug here it snuck into theirs.

henry2man, 2019-11-06:
I think I'm experimenting the same issue here. My non-recurring events are showed correctly but my MM (compiled getting the latest code from the repo) the recurring are showing the end date with one hour offset (my timezone is UTC+1). Also I'm using a iCal from Google Calendar.

Given a non recurring event starting at 17:00 --> MM shows the start date correctly
Given the same event with some recurrency, starting at 17:00 --> MM shows start date as "18:00" (+1 hour).

tisnatch, 2019-11-11:
@nerdoug did your problem recently resolve itself? Mine did. The issue I was having with MMM-CaldendarWeek "fixed itself" this past weekend. All recurring events are now showing up again. I wonder if it's related to the weeks-long period where countries shift off DST.

nerdoug, 2019-11-25:
@tisnatch  Just retested, and still have problems with recurring events during the period when my local date is different than UTC date. In this this test, the event in question appeared 6 days later than it should have - on Dec 2 rather than Nov 26. Events an hour before, and an hour after the "UTC date is different period" worked properly.

tisnatch, 2019-11-26:
So strange. Well, if it helps, I confirm that I _was_ having problems during a one or two week period when the world was shifting off of DST. No doubt, something was screwed up and I tested a few different things. After the DST shift though the dates showed correctly again.

MichMich, 2019-12-29:
Duplicate: https://github.com/MichMich/MagicMirror/issues/1798

closed 2019-12-29 by MichMich