MagicMirrorOrg/MagicMirror #1805: Calendar: showEnd doesn't apply when timeFormat is set to "dateheaders"

jrial, 2019-11-08:
**Platform**:
Linux, Electron 3.0.13

**Node Version**:
v10.17.0

**MagicMirror Version**:
v2

**Description**:
When setting the `timeFormat` to `"dateheaders"`, the `showEnd` setting is ignored. The end times show up when switching to the default `timeFormat`, but disappear when switching back to `"dateheaders"`

**Steps to Reproduce**:
1. Leave `timeFormat` unconfigured; it will default to `"relative"`.
2. Enable end times by adding `showEnd: true,` to the calendar module's config.
3. Reload, verify that end times are indeed shown.
4. Add the line `timeFormat: "dateheaders",` to the config to change the time display to date headers.
5. Reload.

**Expected Results**:
I expect end times to be displayed for all events.

**Actual Results**:
End times are not shown.

**Configuration**:
```
                {
                        module: "calendar",
                        header: "Upcoming Events",
                        position: "top_left",
                        config: {
                                colored: true,
                                coloredSymbolOnly: true,
                                timeFormat: "dateheaders",
                                showEnd: true,
                                fade: false,
                                calendars: [
                                        {
                                                symbol: "calendar-check",
                                                url: "https://XXX.ics"
                                        },
                                        {
                                                symbol: "calendar-week",
                                                url: "https://XXX.ics",
                                                color: "#4b9bcc",
                                        },
                                ]
                        }
                },
```

**Additional Notes**:
N/A

lavolp3, 2019-11-22:
I agree and I even had this programmed a while ago. Need to look if I find it somewhere.
I am using the dateheaders setup and have to say I have my problems with the whole "showEnd" function. Too many possibilities and options to account for.
However I will create a PR to start a discussion about how to implement it.

MichMich, 2019-12-29:
Feel free to send PR.

closed 2019-12-29 by MichMich