MagicMirrorOrg/MagicMirror #1810: Calendar displaying tomorrow all day events, for today, following time change from DST to ST.

scuppasteve, 2019-11-15:
Like it says in the title, i have a calendar for chores that are all day events. It worked fine for months. Following the time change this a couple weekends ago, all the events are one day ahead. For example items for tomorrow are shown today. Below is the config for this module. Any idea?

`		{
			module: "calendar",
			header: "Chores",
			position: "top_right",
			config: {
				fadePoint: 1.0,
				fade: false,
				maximumNumberOfDays: 1,
				maxTitleLength: 40,
				displaySymbol: false,
				getRelative: 0,
				showEnd: false,
				timeFormat: 'absolute',
				calendars: [
					{
						url: 'Calendar URL',
						symbol: 'broom'
					}					
				]
			}
		},`

MichMich, 2019-12-29:
Duplicate.

closed 2019-12-29 by MichMich

hangorazvan, 2019-12-30:
Duplicate with what?
Can you even indicate the post which is duplicated?
Why so much ambiguity is somehow state secret?
The same arrogance as in the forum!

MichMich, 2019-12-30:
https://github.com/MichMich/MagicMirror/issues/1798

You do realize we are all volunteers right?

hangorazvan, 2019-12-30:
Does not matter. With that answer you said everything and solved the problem!