MagicMirrorOrg/MagicMirror #1812: When trying to install NPM I get this error

ECOl10, 2019-11-20:
run-start.sh: 7: run-start.sh: electron: not found

npm ERR! magicmirror@2.9.0 start: `sh run-start.sh`
npm ERR! Exit status 127
npm ERR!
npm ERR! Failed at the magicmirror@2.9.0 start script.
npm ERR! This is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.19.66-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! cwd /home/pi/MagicMirror
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
npm ERR! not ok code 0
pi@raspberrypi:~/MagicMirror $ git pull && npm install
Already up-to-date.
(node:1753) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm WARN engine grunt-markdownlint@2.7.0: wanted: {"node":">=6","npm":">=3"} (current: {"node":"8.11.1","npm":"1.4.21"})
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:120:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.19.66-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/MagicMirror
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
\
> electron-chromedriver@1.8.0 install /home/pi/MagicMirror/node_modules/spectron/node_modules/electron-chromedriver
> node ./download-chromedriver.js

successfully dowloaded and extracted!
npm WARN deprecated left-pad@1.3.0: use String.prototype.padStart()
npm ERR! tar.unpack untar error /home/pi/.npm/postcss-jsx/0.36.3/package.tgz
npm ERR! error rolling back Error: ENOTEMPTY: directory not empty, rmdir '/home/pi/MagicMirror/node_modules/stylelint/node_modules/table/dist'
npm ERR! error rolling back  stylelint@12.0.0 { Error: ENOTEMPTY: directory not empty, rmdir '/home/pi/MagicMirror/node_modules/stylelint/node_modules/table/dist'
npm ERR! error rolling back   errno: -39,
npm ERR! error rolling back   code: 'ENOTEMPTY',
npm ERR! error rolling back   syscall: 'rmdir',
npm ERR! error rolling back   path: '/home/pi/MagicMirror/node_modules/stylelint/node_modules/table/dist' }
npm ERR! tar.unpack untar error /home/pi/.npm/cosmiconfig/6.0.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/global-prefix/3.0.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/loud-rejection/1.6.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/minimist-options/3.0.2/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/camelcase-keys/4.2.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/log-symbols/2.2.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/read-pkg-up/3.0.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/trim-newlines/2.0.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/decamelize-keys/1.1.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/redent/2.0.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/postcss-selector-parser/3.1.1/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/typedarray-to-buffer/3.1.5/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/signal-exit/3.0.2/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/flat-cache/2.0.1/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/remark/10.0.1/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/is-fullwidth-code-point/3.0.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/clone-regexp/2.2.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/braces/3.0.2/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/ansi-regex/5.0.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/unist-util-find-all-after/1.0.5/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/autoprefixer/9.7.2/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/emoji-regex/8.0.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/is-typedarray/1.0.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/chalk/2.4.2/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/picomatch/2.1.1/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/yargs-parser/10.1.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/postcss/7.0.23/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/normalize-package-data/2.5.0/package.tgz
npm ERR! tar.unpack untar error /home/pi/.npm/chalk/2.4.2/package.tgz
npm ERR! error rolling back Error: ENOTEMPTY: directory not empty, rmdir '/home/pi/MagicMirror/node_modules/stylelint/node_modules/chalk/node_modules/supports-color/node_modules/has-flag'
npm ERR! error rolling back  chalk@3.0.0 { Error: ENOTEMPTY: directory not empty, rmdir '/home/pi/MagicMirror/node_modules/stylelint/node_modules/chalk/node_modules/supports-color/node_modules/has-flag'
npm ERR! error rolling back   errno: -39,
npm ERR! error rolling back   code: 'ENOTEMPTY',
npm ERR! error rolling back   syscall: 'rmdir',
npm ERR! error rolling back   path: '/home/pi/MagicMirror/node_modules/stylelint/node_modules/chalk/node_modules/supports-color/node_modules/has-flag' }
npm WARN deprecated core-js@2.6.10: core-js@<3.0 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
\
> core-js@2.6.10 postinstall /home/pi/MagicMirror/node_modules/spectron/node_modules/webdriverio/node_modules/babel-runtime/node_modules/core-js
> node postinstall || echo "ignore"

npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
npm ERR! not ok code 0

lavolp3, 2019-11-22:
Not sure if it's the reason but your npm is very very outdated.
Try to do 
`sudo npm install npm@latest -g`
and then npm install again in the MM folder.

sdetweil, 2019-11-22:
why are u doing this install manually?  the script does all the work.

ercol10, 2019-11-27:
after updating and installing npm again this is now tthe new error message that i am getting 

> magicmirror@2.9.0 start /home/pi/MagicMirror
> sh run-start.sh

run-start.sh: 7: run-start.sh: electron: not found
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! magicmirror@2.9.0 start: `sh run-start.sh`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the magicmirror@2.9.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-11-27T04_39_03_471Z-debug.log

sdetweil, 2019-11-27:
tell me more.. how did you update/install?

ECOl10, 2019-12-04:
this is a list of everything that I ran after running   " sudo npm install npm@latest -g "
 -" get update - index "
 -" node install. js "
 -" sudo npm install npm -g "

sdetweil, 2019-12-04:
please use the new installer,  see the forum topic here for the link
https://forum.magicmirror.builders/topic/10171/anyone-want-to-try-updated-installer

will have to rename the MagicMirror folder out of the way

ercol10, 2019-12-04:
thanks for the help on this ! I fixed this error !

sdetweil, 2019-12-04:
what did u do?

ercol10, 2019-12-04:
I ran git pull && npm install inside of the MagicMirror File

sdetweil, 2019-12-04:
can I ask why you didn't use the automated installer? (either one?)

ECOl10, 2019-12-04:
I tired installing manually then deleted the everything and started over once I found the automated installed. The errors that I am getting is from the automated installer

ECOl10, 2019-12-04:
I tired installing manually then deleted the everything and started over once I found the automated installed. The errors that I am getting are from the automated installer

sdetweil, 2019-12-04:
Thanks. Can u say which install script you used?

sdetweil, 2019-12-07:
if you are using my updated script, please post the ~/MagicMirror/install.log

MichMich, 2019-12-29:
Can this be closed?

sdetweil, 2019-12-29:
i think so, this was due to the downlevel npm (npm ERR! npm -v 1.4.21) because they split the node and npm installs apart, and the current installer (2.9) doesn't handle that

closed 2019-12-29 by MichMich