MagicMirrorOrg/MagicMirror #1818: Rotating cause black screen after restart

stephane-r, 2019-12-01:
Hi community !

I try to rotate my screen with fresh rasbian/MM installation. But for me, it show black screen. You can see the detail :

**Platform**: Raspberry Pi 3

**Node Version**: 10.x

**MagicMirror Version**: Latest

**Description**: I try 

**Steps to Reproduce**: I did all step for rotate screen on the documentation (https://github.com/MichMich/MagicMirror/wiki/Configuring-the-Raspberry-Pi).

**Expected Results**: After rasbian boot (the rasbian screen on starting), the screen is only black, no desktop displayed, but SSH work.

Anyone have already rotate screen on latest rasbian/MM ?

Thank you !

emilb, 2019-12-02:
I can confirm the described behaviour. Screen is just black after rotating the screen. I used the following settings in /boot/config.txt:
dtoverlay=vc4-fkms-v3d
display_rotate=1
avoid_warnings=1

Some text is shown if I use the mouse to select stuff. Lots of the modules are shown in the wrong place.

emilb, 2019-12-02:
I got it working on my Rpi 3B by changing my /boot/config.txt to:

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1
display_rotate=1
avoid_warnings=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0

gpu_mem=256
# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

closed 2019-12-29 by MichMich