MagicMirrorOrg/MagicMirror #1820: Description of MM page

skrutt, 2019-12-03:
* A short, one sentence, clear description what it does (duh!)

This is not present on the magic mirror page [https://magicmirror.builders/](https://magicmirror.builders/)

And frankly, i would still like to know what it does? There is a video, but that's not really compatible with my current office situation.

Just my 2 cent, sorry for posting this at what is likely the wrong place but hopefully you know the correct contacts anyway :)

sdetweil, 2019-12-03:
from the github repo 

MagicMirror² is an open source modular smart mirror platform. With a growing list of installable modules, the MagicMirror² allows you to convert your hallway or bathroom mirror into your personal assistant.

skrutt, 2019-12-03:
Yes, thanks, But what does it mean? What use do i have of installing modules into my what? From the page, imho, you can derive that what the thing is, probably is software. But could also be hardware? 

which is my point, it sounds lovely, it's fun to see everyones excitement, but what is it?

Maybe it's not meant as the users first contact with the concept smart mirror, but i dont see why it could not be. 

Again, stop me if i'm rambling, i know im a bit obtuse at times, but a bit more descriptive intro would be helpful to actually understand what this platform actually does... "Unfortunately, more often than not, this information is rarely available nor easily obtained without having to install it."

(again, maybe it's crystal in the video, but that's not an option for me atm.. (also, the blog does not show correctly here so cant look there either))

sdetweil, 2019-12-03:
the key term is 'smart mirror'

given a short search 

A smart mirror is a device that functions as a mirror with additional capability of displaying multimedia data, such as text, images, and videos. This device allows users to access and interact with contextual information, such as weather data, seamlessly as part of their daily routine

there is a general assumption that if the reader doesn't understand a term, they will research it. 

as for hardware/software..

this is the SOFTWARE site, github. so this is software.  but software doesn't run by itself.. it needs a computer hardware instance... the doc says raspberry pi.. (another term needs to be looked up maybe). 

what else does it take?  well, thats up to the reader... by going to the referenced site, you will be exposed to other users 'use cases' and implementations. 

so, given that short bit of info, do you want or need a computer (monitor/display) to show you info (weather, clock, calendar ...etc) in a continuous way . if the answer is yes, then this MIGHT be a possible way to construct such a device with a presentation of your liking

skrutt, 2019-12-03:
"A smart mirror is a device that functions as a mirror with additional capability of displaying multimedia data, such as text, images, and videos. This device allows users to access and interact with contextual information, such as weather data, seamlessly as part of their daily routine"

Perfect! put that on https://magicmirror.builders/ and call it a day :)

And yes, github is a software site.. well, almost at least, but https://magicmirror.builders/ is just a site for the MM platform, that (in my opinion) should have a clear description in text.

If we play with that this was a hugely popular idea and MM was a very successful company, in fact it was the first hit on smart mirror after a google search. Would you not expect there to be a textual description of this concept there? And even if this is not a company and not the number one google hit for 'smart mirror', why not strive to be equally informative and clear?

sdetweil, 2019-12-03:
thanks.. I am just another user.

i was and am satisfied with the description.  this is an opensource project, started by one guy. not a company.  all volunteer

closed 2019-12-29 by MichMich