MagicMirrorOrg/MagicMirror #1821: Access denied to IP address in ipWhitelist

fire1ce, 2019-12-03:
config:
var config = {
    address: '0.0.0.0', // Address to listen on, can be:
    // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
    // - another specific IPv4/6 to listen on a specific interface
    // - "", "0.0.0.0", "::" to listen on any interface
    // Default, when address config is left out, is "localhost"
    port: 8080,
    ipWhitelist: [
        '127.0.0.1',
        '::ffff:127.0.0.1',
        '::1',
        '::ffff:192.168.1.1/120',
        '::ffff:172.16.1.1/120'
    ],

Log: 
0|mmm    | > magicmirror@2.9.0 start /Users/mmm/MagicMirror
0|mmm    | > sh run-start.sh
0|mmm    | cut: illegal option -- -
0|mmm    | usage: cut -b list [-n] [file ...]
0|mmm    |        cut -c list [file ...]
0|mmm    |        cut -f list [-s] [-d delim] [file ...]
0|mmm    | Starting MagicMirror: v2.9.0
0|mmm    | Loading config ...
0|mmm    | Loading module helpers ...
0|mmm    | No helper found for module: alert.
0|mmm    | Initializing new module helper ...
0|mmm    | Module helper loaded: updatenotification
0|mmm    | No helper found for module: clock.
0|mmm    | Initializing new module helper ...
0|mmm    | Module helper loaded: calendar
0|mmm    | No helper found for module: MMM-WiFiPassword.
0|mmm    | No helper found for module: compliments.
0|mmm    | Initializing new module helper ...
0|mmm    | Module helper loaded: MMM-DarkSkyForecast
0|mmm    | Initializing new module helper ...
0|mmm    | Module helper loaded: newsfeed
0|mmm    | All module helpers loaded.
0|mmm    | Starting server on port 8080 ...
0|mmm    | Server started ...
0|mmm    | Connecting socket for: updatenotification
0|mmm    | Connecting socket for: calendar
0|mmm    | Starting node helper for: calendar
0|mmm    | Connecting socket for: MMM-DarkSkyForecast
0|mmm    | ====================== Starting node_helper for module [MMM-DarkSkyForecast]
0|mmm    | Connecting socket for: newsfeed
0|mmm    | Starting module: newsfeed
0|mmm    | Sockets connected & modules started ...
0|mmm    | 2019-12-03 22:00:10.244 Electron[574:6716] ApplePersistence=NO
0|mmm    | Launching application.
0|mmm    | Create new calendar fetcher for url: http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics - Interval: 300000
0|mmm    | Create new news fetcher for url: http://www.ynet.co.il/Integration/StoryRss1854.xml - Interval: 300000
0|mmm    | Access denied to IP address: 192.168.1.125
0|mmm    | Access denied to IP address: 192.168.1.125

sdetweil, 2019-12-04:
that 192.168.1.1/120 is an invalid address mask.. 

120 can be at MAX 32 (all 32 bits of address allowed)

you could do 192.168.1.125/32 to ONLY allow from that address

fire1ce, 2019-12-04:
> that 192.168.1.1/120 is an invalid address mask..
> 
> 120 can be at MAX 32 (all 32 bits of address allowed)
> 
> you could do 192.168.1.125/32 to ONLY allow from that address

I would like to enable whitelist my local pull and my vpn pull. what's the right syntax to do this?

fire1ce, 2019-12-04:
found the right syntax:

> ipWhitelist: [
>         '127.0.0.1',
>         '::ffff:127.0.0.1',
>         '::1',
>         '192.168.1.0/24',
>         '172.16.1.0/24'
>     ],

closed 2019-12-04 by fire1ce