MagicMirrorOrg/MagicMirror #1824: Error when running pm2 start mm.sh

rsperezn, 2019-12-04:
Hi guys,
I just installed MM on my raspberry pi via the [Automatic Installation](https://github.com/MichMich/MagicMirror#manual-installation) and it started fine. Aftewards I had  to restart my pi and wanted to get MM running so I followed the guide on [autostarting](https://github.com/MichMich/MagicMirror/wiki/auto-starting-magicmirror) the mirror but I'm getting the following error when running _any_ `pm2` command.
```/usr/local/lib/node_modules/pm2/node_modules/needle/lib/multipart.js:1
(function (exports, require, module, __filename, __dirname) { ���������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������

SyntaxError: Invalid or unexpected token
    at new Script (vm.js:79:7)
    at createScript (vm.js:251:10)
    at Object.runInThisContext (vm.js:303:10)
    at Module._compile (internal/modules/cjs/loader.js:657:28)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
```
If if run `npm start` the mirror starts fine.
This is my configuration
raspberry pi 3 model b+
npm -version 6.13.2
node --version v10.15.2
magicmirror@2.9.0 
Any clues?

sdetweil, 2019-12-07:
did u have pm2 installed before?  usually this happens when the npm runtime changes out from under pm2. 
if you use my script, it will uninstall/reinstall pm2 and do the setup

see this topic 
https://forum.magicmirror.builders/topic/10907/new-script-to-install-fix-pm2-config-for-start-on-boot

rsperezn, 2019-12-08:
I don't think I had pm2 since it was a  was a fresh install of Buster. Nonetheless, after running your script it works perfectly :) thanks for the help

closed 2019-12-08 by rsperezn