MagicMirrorOrg/MagicMirror #1830: Date is off by 24 hr

aa0pb, 2019-12-12:
Raspberry Pi 3
Node: v10.17.0
MagicMirror Version: 2

**Description**: 
Howdy,
Yes I have search here and Google and no the “showEnd: false” does not work.
It has been doing this for a while on two units and I just did a fresh install on one so new setup.

Anyone else still having an issue? The dates are show one day before the time on the calendar. And it is doing it on new and old entries.

Thanks for any help.

**Steps to Reproduce**: Basic setup to Google Calendar
**Expected Results**: To see the correct times/dates
**Actual Results**: One day early
**Configuration**: 

 {
                        module: "calendar",
                        header: "Family Calendar",
                        position: "top_left",
                        pages: {"main": "top_left"},
                        config: {
                                maximumEntries: 6,
                                dateFormat: 'D MMMM - H:mm',
                                fullDayEventDateFormat: 'D MMMM',
                                showEnd: false,
                                calendars: [
                                        {
                                                symbol: "calendar-alt",
                                                url: "GoogleLink"
                                        }
                                ]
                        }
                },

Ybbet, 2019-12-13:
Hi. 
I've got a similar problem. I use multiple calendars from Google.
The problem I have is for events without a schedule, an event of the day. I use this type of event for household chores. When I look at the ics file, it shows me these events with an offset. Google Calendar puts me start times in the middle of the day (ex: 9:59 am) and not at midnight (0:00 am to 11:59 pm). MagicMirror detects that this is a day event. I think the error is not from Magic Mirror but from Google Calendar.

aa0pb, 2019-12-16:
Ybbet, thanks for the note to look at the ics file.  I missed it before, but not all the dates are wrong.

If the ICS event start/end field has the TZ in the field, then it is correct.
If there is no TZ listed the time is listed 23 hrs before the actual start time.  There is no all day events.
I am looking at the Google calendar to see if there is a way to force the TZ in the event.
This is on both reoccurring events and single events also.

MichMich, 2019-12-29:
Duplicate.

closed 2019-12-29 by MichMich

aa0pb, 2019-12-30:
If it is duplicate please supply the link to the dup?