MagicMirrorOrg/MagicMirror #1833: Symbol?

GoodWillGustin, 2019-12-19:
To change the 'symbol' icon, do we need to download the icons from the site you mentioned? I have set several different icons from the names on http://fontawesome.io/icons/ -- but the icons just show as empty boxes. Am I doing something wrong?

sdetweil, 2019-12-24:
what module is this in? weather?

GoodWillGustin, 2019-12-27:
Sorry, I think I may have posted this in the wrong place somehow. I was referring to the 'calendar' module. 

https://github.com/MichMich/MagicMirror/tree/master/modules/default/calendar

GoodWillGustin, 2019-12-27:
The 'calendar' module.

https://github.com/MichMich/MagicMirror/tree/master/modules/default/calendar

sdetweil, 2019-12-27:
code uses the "font-awesome.css".. I don't know what symbols are supported by what css files..
(I am just another user like you)

closed 2019-12-29 by MichMich