MagicMirrorOrg/MagicMirror #1840: Issues with Weather Tests.

MichMich, 2019-12-29:
More info soon.

roramirez, 2019-12-31:
After merge #1849 I could check it

MichMich, 2020-01-02:
Fixed by reverting electron. See https://github.com/MichMich/MagicMirror/issues/1800

closed 2020-01-02 by MichMich

sdetweil, 2020-01-03:
still failing today

MichMich, 2020-01-03:
They run ok now. They timeout after 10sec. We might need to stretch that 10 sec.

sdetweil, 2020-01-03:
The code says if under CI then timeout is 30 seconds

MichMich, 2020-01-03:
This line (for example) uses 10000ms:
https://github.com/MichMich/MagicMirror/blob/69a887fb05af79071eb20875c1c4a856627f7811/tests/e2e/modules/weather_spec.js#L48

sdetweil, 2020-01-03:
Ok. Other tests call a function in global_setup to set timeouts. Figured they all would

roramirez, 2020-01-21:
I've try catch the failed of this and  apparently is because is too slow to run these tests. There a other line
https://github.com/MichMich/MagicMirror/blob/69a887fb05af79071eb20875c1c4a856627f7811/tests/e2e/modules/weather_spec.js#L11
for a wait. Maybe @fewieden can check a look of this.

fewieden, 2020-01-22:
@roramirez @MichMich I tried this morning the newest version of develop and run the tests locally, they all passed after ~7s.

What is the setup I need to reproduce the issue?

Since this seems to be still relevant, we should reopen the issue.

roramirez, 2020-01-22:
Hi @fewieden 

You should try into a similar environment where is running like Travis. Maybe you can using the follow guide (I'm use this one) https://jonlabelle.com/snippets/view/markdown/run-travis-ci-builds-locally-with-docker.

Or also requests the debug mode for your Fork in Travis 
https://docs.travis-ci.com/user/running-build-in-debug-mode/#enabling-debug-mode

> Since this seems to be still relevant, we should reopen the issue.

Yes. Should reopen this issue

roramirez, 2020-01-22:
@fewieden I forget other thing. Also should run the test against the PR https://github.com/MichMich/MagicMirror/issues/1840

fewieden, 2020-02-02:
@roramirez the PR that you tried to reference is a link to this issue.

Therefore I tried https://jonlabelle.com/snippets/view/markdown/run-travis-ci-builds-locally-with-docker to debug this [commit](https://github.com/MichMich/MagicMirror/commit/c1e808bce66ec1d8e937f5dac91f2ca6f695e8de) which had the original build [log](https://travis-ci.com/MichMich/MagicMirror/builds/142541536).

Executing the tests inside the docker container resulted in a different error

[image]

What I found is https://stackoverflow.com/questions/50642308/webdriverexception-unknown-error-devtoolsactiveport-file-doesnt-exist-while-t but even with adding all those argumens to the chrome driver the issue still appeared. Maybe I passed them wrong?

Running the tests locally gives a third result. Sometimes they pass, sometimes the same tests fail. I could see that they made actual requests to the api server instead of using the mock. Maybe we have a timing iue in there.

Updating electron and spectron to their latest stable versions is passing all the [tests](https://travis-ci.com/fewieden/MagicMirror/builds/147096746). What is failing, is the npm list command afterwards.

[image]

CC @MichMich

roramirez, 2020-02-05:
Hi @fewieden You're right. I've reference wrong the  PR. This should be #1892. Sorry for that. This change is already merge in `develop` you could try against this branch. The test for weather is disabled in this branch.

The problem you are getting is because the sandbox mode is enable by default. More information about this is  #1892

Setting  the enviroment variable `ELECTRON_DISABLE_SANDBOX=1`  before run the test should be enough.

fewieden, 2020-02-10:
@roramirez looks like I found the issue

roramirez, 2020-02-12:
@fewieden Nice work!. Seems there this was little trickle but you found it.

@MichMich You should merge the #1927

MichMich, 2020-02-12:
Merged!