MagicMirrorOrg/MagicMirror #1842: Issues with Vendor_spec tests (e2e)

sdetweil, 2019-12-29:
when running the automated tests, in the E2E section, there is a test to insure download of the vendor provided javascript files. 

each access is returning  ECONNREFUSED 127.0.0.1:8080
but that should be correct and does work when manually executed (via curl or wget)

there are two sections of the test... get from correct path
and get from bad path... 

both sections fail the same
but report passed

v 2.9.0 or above

reproduce
install selected version, npm install
then run 
npm run test:e2e

[screen shot](https://www.dropbox.com/s/08eq118ez6b29cf/2019-12-29%20%281%29.png?dl=0)

MichMich, 2019-12-29:
Just to be sure: the test passes, but the test isn't correct?

sdetweil, 2019-12-29:
the test does not fail, but the actual items to be tested are not actually tested

sdetweil, 2020-01-08:
code changes, vendor.js
```
	before(function () {
		console.log("starting js/electron");
		return helpers.startApplication({
			args: [__dirname + "/../../js/electron.js"]
		}).then(function (startedApp) { app = startedApp; console.log("parms="+JSON.stringify(app.getSettings()))});  // dump settings after app created
	});

				urlVendor = "http://localhost:9515/vendor/" + vendors[vendor];
```
output
```
starting js/electron
parms={"host":"127.0.0.1",**"port":9515**,"quitTimeout":1000,"startTimeout":5000,"waitTimeout":5000,"connectionRetryCount":10,"connectionRetryTimeout":30000,"nodePath":"/usr/bin/node","path":"/home/sam/MagicMirror/node_modules/.bin/electron","args":["/home/sam/MagicMirror/tests/e2e/../../js/electron.js"],"chromeDriverArgs":[],"env":{},"workingDirectory":"/home/sam/MagicMirror","webdriverOptions":{},"requireName":"require"}
    Get list vendors
      ✓ should return 200 HTTP code for vendor "moment.js"
      ✓ should return 200 HTTP code for vendor "moment-timezone.js"
      ✓ should return 200 HTTP code for vendor "weather-icons.css"
      ✓ should return 200 HTTP code for vendor "weather-icons-wind.css"
      ✓ should return 200 HTTP code for vendor "font-awesome.css"
      ✓ should return 200 HTTP code for vendor "nunjucks.js"
      ✓ should return 200 HTTP code for vendor "suncalc.js"
      ✓ should return 404 HTTP code for vendor https://localhost/"moment.js"
      ✓ should return 404 HTTP code for vendor https://localhost/"moment-timezone.js"
      ✓ should return 404 HTTP code for vendor https://localhost/"weather-icons.css"
      ✓ should return 404 HTTP code for vendor https://localhost/"weather-icons-wind.css"
      ✓ should return 404 HTTP code for vendor https://localhost/"font-awesome.css"
      ✓ should return 404 HTTP code for vendor https://localhost/"nunjucks.js"
      ✓ should return 404 HTTP code for vendor https://localhost/"suncalc.js"
```

sdetweil, 2020-01-08:
here is what the current test results look like,  the ECONNREFUSED, is typical of not connecting on the correct port
```
    Get list vendors
      ✓ should return 200 HTTP code for vendor "moment.js"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "moment-timezone.js"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "weather-icons.css"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "weather-icons-wind.css"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "font-awesome.css"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "nunjucks.js"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 200 HTTP code for vendor "suncalc.js"
    - There error vendor 200 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"moment.js"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"moment-timezone.js"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"weather-icons.css"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"weather-icons-wind.css"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"font-awesome.css"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"nunjucks.js"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
      ✓ should return 404 HTTP code for vendor https://localhost/"suncalc.js"
    - There error vendor 404 test Error: connect ECONNREFUSED 127.0.0.1:8080
```

roramirez, 2020-01-22:
@sdetweil The PR #1893 fix this situation. Very thanks for all feedback about this.

roramirez, 2020-01-22:
The PR is already merged. We'll should close this issue

cc @sdetweil @MichMich

khassel, 2020-01-22:
one question @roramirez: Why are such errors treated as `pending` and not as `error`?

roramirez, 2020-01-23:
> one question @roramirez: Why are such errors treated as `pending` and not as `error`?

Because is to show as warning to know what string are missing. If we treated as error the test CI process will fail until the all string are translated.

khassel, 2020-01-23:
thats clear for the translation strings, but the question was about the vendor tests.

sdetweil, 2020-01-24:
i think the testcase is written  incorrectly

it should signal completed like the others
```
			it(`should return 200 HTTP code for vendor "${vendor}"`, function (done) {
				urlVendor = "http://localhost:8080/vendor/" + vendors[vendor];
				request.get(urlVendor, function (err, res, body) {
					if (!err){
						expect(res.statusCode).to.equal(200);
						done();
					}
```

same for the expected failing testcases

roramirez, 2020-01-25:
@khassel You're right!, I confussing about the issue!

@sdetweil Yes, should be write this way. We've introduce this change in 1e97b5c27a1275bee56335d9fa657bc1dbac7a6b time ago because the CI failure randomize way. To prevent the failed we set as error log instead of raise failed for this test. 

I pushed the #1903 to change the expectation of results for this test.

sdetweil, 2020-01-25:
@roramirez 

note that you have to pass the done method in 

it(`should return 200 HTTP code for vendor "${vendor}"`, function (done) {   <----- 

else it fails as in the test results (done not found)

roramirez, 2020-01-26:
@sdetweil Thanks for notice!. I've done the change for the commit and I removed the `done`. I was thinking and  we not should keep this for  the test is using a iterator.

sdetweil, 2020-01-26:
so what?  it is calling this for EACH TEST, so done is the 'clean'  way to single this test is complete (for each iteration)

it works properly   add the done parm to the  function invocation
and then use it in the function...

roramirez, 2020-01-27:
The PR is already merge into `develop` branch. Maybe you can add this behavior. 

This issue should close.

sdetweil, 2020-01-27:
@roramirez  cool, but the PR is in error...  why should I have to fix it?

roramirez, 2020-01-27:
I mean add the `done` as you're suggested before

closed 2020-04-03 by stale[bot]