MagicMirrorOrg/MagicMirror #1843: Base translations have fallen behind newer versions

sdetweil, 2019-12-29:
when running the automated tests, e2e, there is a section to verify that all the used terms are translated in all the supported languages,

this test now reports that there are many missing translations
```
  Translation keys of nl
        ✓ nl keys should be in base
    - Missing Translations for language nl: PRECIP,WEEK
        - nl should contain all base keys
      Translation keys of de
        ✓ de keys should be in base
    - Missing Translations for language de: PRECIP
        - de should contain all base keys
      Translation keys of fi
        ✓ fi keys should be in base
    - Missing Translations for language fi: PRECIP,UPDATE_INFO_MULTIPLE,UPDATE_INFO_SINGLE
        - fi should contain all base keys
      Translation keys of fr
        ✓ fr keys should be in base
    - Missing Translations for language fr: PRECIP
        - fr should contain all base keys
      Translation keys of fy
        ✓ fy keys should be in base
    - Missing Translations for language fy: FEELS,PRECIP,UPDATE_INFO_MULTIPLE,UPDATE_INFO_SINGLE,UPDATE_NOTIFICATION,UPDATE_NOTIFICATION_MODULE,WEEK
        - fy should contain all base keys
      Translation keys of es
        ✓ es keys should be in base
        ✓ es should contain all base keys
      Translation keys of ca
        ✓ ca keys should be in base
    - Missing Translations for language ca: FEELS,PRECIP
        - ca should contain all base keys
      Translation keys of nb
        ✓ nb keys should be in base
    - Missing Translations for language nb: PRECIP
        - nb should contain all base keys
      Translation keys of nn
        ✓ nn keys should be in base
    - Missing Translations for language nn: PRECIP,WEEK
        - nn should contain all base keys
      Translation keys of pt
        ✓ pt keys should be in base
    - Missing Translations for language pt: PRECIP
        - pt should contain all base keys
      Translation keys of pt-br
        ✓ pt-br keys should be in base
    - Missing Translations for language pt-br: DAYAFTERTOMORROW,FEELS,PRECIP,WEEK
        - pt-br should contain all base keys
      Translation keys of sv
        ✓ sv keys should be in base
    - Missing Translations for language sv: PRECIP
        - sv should contain all base keys
      Translation keys of id
        ✓ id keys should be in base
    - Missing Translations for language id: FEELS,PRECIP
        - id should contain all base keys
      Translation keys of it
        ✓ it keys should be in base
    - Missing Translations for language it: PRECIP
        - it should contain all base keys
      Translation keys of zh-cn
        ✓ zh-cn keys should be in base
    - Missing Translations for language zh-cn: PRECIP
        - zh-cn should contain all base keys
      Translation keys of zh-tw
        ✓ zh-tw keys should be in base
    - Missing Translations for language zh-tw: DAYAFTERTOMORROW,FEELS,PRECIP,UPDATE_INFO_MULTIPLE,UPDATE_INFO_SINGLE,UPDATE_NOTIFICATION,UPDATE_NOTIFICATION_MODULE,WEEK
        - zh-tw should contain all base keys
      Translation keys of ja
        ✓ ja keys should be in base
    - Missing Translations for language ja: DAYAFTERTOMORROW,FEELS,PRECIP,UPDATE_INFO_MULTIPLE,UPDATE_INFO_SINGLE,UPDATE_NOTIFICATION,UPDATE_NOTIFICATION_MODULE,WEEK
        - ja should contain all base keys
      Translation keys of pl
        ✓ pl keys should be in base
    - Missing Translations for language pl: PRECIP
        - pl should contain all base keys
      Translation keys of gr
        ✓ gr keys should be in base
    - Missing Translations for language gr: DAYAFTERTOMORROW,FEELS,PRECIP,UPDATE_INFO_MULTIPLE,UPDATE_INFO_SINGLE,UPDATE_NOTIFICATION,UPDATE_NOTIFICATION_MODULE,WEEK
        - gr should contain all base keys
      Translation keys of da
        ✓ da keys should be in base
    - Missing Translations for language da: PRECIP
        - da should contain all base keys
      Translation keys of tr
        ✓ tr keys should be in base
    - Missing Translations for language tr: DAYAFTERTOMORROW,FEELS,PRECIP,UPDATE_INFO_MULTIPLE,UPDATE_INFO_SINGLE,UPDATE_NOTIFICATION,UPDATE_NOTIFICATION_MODULE,WEEK
        - tr should contain all base keys
      Translation keys of ru
        ✓ ru keys should be in base
    - Missing Translations for language ru: PRECIP
        - ru should contain all base keys
      Translation keys of af
        ✓ af keys should be in base
    - Missing Translations for language af: FEELS,PRECIP,WEEK
        - af should contain all base keys
      Translation keys of hu
        ✓ hu keys should be in base
    - Missing Translations for language hu: PRECIP
        - hu should contain all base keys
      Translation keys of is
        ✓ is keys should be in base
    - Missing Translations for language is: FEELS,PRECIP,WEEK
        - is should contain all base keys
      Translation keys of et
        ✓ et keys should be in base
    - Missing Translations for language et: FEELS,PRECIP,WEEK
        - et should contain all base keys
      Translation keys of kr
        ✓ kr keys should be in base
    - Missing Translations for language kr: FEELS,PRECIP,WEEK
        - kr should contain all base keys
      Translation keys of ro
        ✓ ro keys should be in base
    - Missing Translations for language ro: PRECIP
        - ro should contain all base keys
      Translation keys of cy
        ✓ cy keys should be in base
    - Missing Translations for language cy: FEELS,PRECIP
        - cy should contain all base keys
      Translation keys of bg
        ✓ bg keys should be in base
    - Missing Translations for language bg: FEELS,PRECIP
        - bg should contain all base keys
      Translation keys of cs
        ✓ cs keys should be in base
    - Missing Translations for language cs: FEELS,PRECIP
        - cs should contain all base keys
      Translation keys of hr
        ✓ hr keys should be in base
    - Missing Translations for language hr: PRECIP
        - hr should contain all base keys
      Translation keys of sk
        ✓ sk keys should be in base
    - Missing Translations for language sk: FEELS,PRECIP
        - sk should contain all base keys
      Translation keys of tlh
        ✓ tlh keys should be in base
    - Missing Translations for language tlh: PRECIP
        - tlh should contain all base keys
      Translation keys of ms-my
        ✓ ms-my keys should be in base
    - Missing Translations for language ms-my: FEELS,PRECIP
        - ms-my should contain all base keys
```

we should solicit help to get these updated, else remove the test

MichMich, 2019-12-29:
It seems something else is wrong. Every time a different test is failing.

sdetweil, 2019-12-29:
no.. all of these fail together.. i thought, after u posted the 1st, that I should create issues for the others so we fix them..  everything else passes

sdetweil, 2019-12-29:
i have opened a topic on the forum 
https://forum.magicmirror.builders/topic/11872/need-help-default-translations-out-of-date
to solicit help

MichMich, 2019-12-29:
I think we should change the tests. There is a fallback to English, so no need for failed tests.

sdetweil, 2019-12-29:
your call. i think we will get forum topics for incomplete translations.. you still have a large wave of new users coming on.. 

these aren't fails', these are 'warnings'..

sdetweil, 2019-12-29:
one thing to note is that the travis tests are running on trusty 
.travis.yml
dist: trusty

I have not checked all the devdependencies for updates.. I know I had to update 1 of them due to the electron version

roramirez, 2019-12-31:
The failing of the test is by other reason. The test about translations is a good picture for are missing strings, that is the sense of this test. 

About the failures please check it the #1849

sdetweil, 2019-12-31:
this one is ok..  it identifies things we need to do...

ZakarFin, 2020-01-01:
I can do a PR for the finnish ones. Whats the meaning of PRECIP? It's PoP in english but I don't have a clue about the context for translation.

sdetweil, 2020-01-01:
Probability of rain/snow

sdetweil, 2020-01-01:
The French abbrev is Pdp

ZakarFin, 2020-01-01:
I don't think we have any acronym for PoP in Finnish. I found one paper that referenced it as P<sub>rain</sub> and it's "Sateen todennäköisyys" when spelled out. Is it too long for the UI? Didn't see it in the UI after updating.

sdetweil, 2020-01-01:
Ok. Thanks for helping. Really appreciate your time

roramirez, 2020-01-22:
Why not do a ping a to translations contributors for the missing strings?

sdetweil, 2020-01-22:
already done
see https://forum.magicmirror.builders/topic/11872/need-help-default-translations-out-of-date

roramirez, 2020-01-22:
Nice! should keep open this issue at the new release?, I don't think we need keeping this open.

sdetweil, 2020-01-22:
i think we should keep AN issue open for the PR's to reference

closed 2020-03-29 by stale[bot]