MagicMirrorOrg/MagicMirror #1852: Version 2.10.0 breaks serveronly mode for remote hosts

HeikoGr, 2020-01-02:
**Platform**: 
Raspberry Pi 4 as Server
Raspberry Pi Zero as Display

**Node Version**:
v13.5.0

**MagicMirror Version**:
2.10.0

**Description**: 
I run MM² serveronly on a headless RPi4.
After the upgrade to version 2.10.0 the display (on remote host) was unable to connect to the server.

The Server itself could show the MagicMirror when used by https://localhost:8080. NOT if used by ip adess 192.168.0.12:80

**Configuration**: 
```
var config = {
    adress: "",
```

**Steps to Reproduce**: 
see description

**Additional Notes**: 
I found a Workaround by reverting commit ed61ac6 and change line 27 back to:
server.listen(port, config.address ? config.address : null);

HeikoGr, 2020-01-02:
Or i use:
`var config = {
    adress: "",`

an let commit ed61ac6 unchanged.

so you may update the config sample not to use `"" `anymore.

sdetweil, 2020-01-02:
according to the doc for magicmirror, if the address field is left empty or not specified, 'localhost' will be used..  NOT '0.0.0.0'

```
The interface ip address on which to accept connections. The default is localhost, which would prevent exposing the built-in webserver to machines on the local network. To expose it to other machines, use: 0.0.0.0.
```

HeikoGr, 2020-01-02:
according to the inline doc (config.js.sample):

```
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "", "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out, is "localhost"
```

sdetweil, 2020-01-02:
I didn't read the file,  only the website doc

https://github.com/MichMich/MagicMirror#general

see the address field

HeikoGr, 2020-01-02:
i saw it after your comment :-D

To be pedantic the "" case is not explained in "README.md" (but in config/config.js.sample)
So your commit did break a working and allowed (well ... not forbidden) config variant.

Can you commit a patch either correcting

- config/config.js.sample
- README.md
- js/server.js

because i am a total fool in github...

sdetweil, 2020-01-02:
>because i am a total fool in github...

I will teach you

on the master repo, on github, upper right, click fork.
now YOU have an editable copy of the repo
you can clone YOUR fork

edit the files.
add a line to the top of CONTRIBUTING.md
to describe your change

git status will show they changed
git add ???
for each ??? =filename

git commit -m " some message"  
you will have to add your name and email before the commit will work
the commit again

once commit is done,
git push
to copy the changes to YOUR github repo
now on github, you should see that there is a change, push create pull request
fill in the title and content
also reference this issue number #number
and submit!

now YOU have contributed to the MM project

HeikoGr, 2020-01-02:
#1857 Pull Request sent (thanks @sdetweil)

sdetweil, 2020-01-02:
can u add a line to the changelog file?   I gave the wrong filename..

CHANGELOG.md

MichMich, 2020-01-02:
Fix merged.

closed 2020-01-02 by MichMich