MagicMirrorOrg/MagicMirror #1853: MM fails to start in v2.10.0 on raspberry pi

rsperezn, 2020-01-02:
Hey guys,
Just upgraded to the latest version `v2.10.0` and  MM does not seem to start. :grimacing: 
I run ```pm2 start mm.sh 
[PM2] Starting /home/pi/MagicMirror/mm.sh in fork_mode (1 instance)
[PM2] Done.
┌────┬────────────────────┬──────────┬──────┬───────────┬──────────┬──────────┐
│ id │ name               │ mode     │ ↺    │ status    │ cpu      │ memory   │
├────┼────────────────────┼──────────┼──────┼───────────┼──────────┼──────────┤
│ 0  │ mm                 │ fork     │ 0    │ online    │ 0%       │ 2.5mb    │
│ 1  │ mm                 │ fork     │ 0    │ online    │ 0%       │ 2.4mb    │
└────┴────────────────────┴──────────┴──────┴───────────┴──────────┴──────────┘
[PM2][WARN] Current process list running is not in sync with saved list. Type 'pm2 save' to synchronize or enable autosync via 'pm2 set pm2:autodump true' ``` where the contents of mm.sh is as described [here](https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror). I tried running `pm2 save` as printed in the terminal but did not help. Also tried running [this](https://forum.magicmirror.builders/topic/10907/new-script-to-install-fix-pm2-config-for-start-on-boot) as suggested by @sdetweil as a response to #1824, however it does not work either.

I tried also run `./run-start.sh`  but then I get  `./run-start.sh: line 66: electron: command not found`. Finally running  `pm2 logs mm` I can see 
``` [TAILING] Tailing last 15 lines for [mm] process (change the value with --lines option)
/home/pi/.pm2/logs/mm-out.log last 15 lines:
0|mm       | 
0|mm       | > magicmirror@2.10.0 start /home/pi/MagicMirror
0|mm       | > ./run-start.sh
0|mm       | 
0|mm       | 
0|mm       | > magicmirror@2.10.0 start /home/pi/MagicMirror
0|mm       | > ./run-start.sh
0|mm       | 

/home/pi/.pm2/logs/mm-error.log last 15 lines:
0|mm       |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
0|mm       |     at Module.load (internal/modules/cjs/loader.js:620:32)
0|mm       | A JavaScript error occurred in the main process
0|mm       | Uncaught Exception:
0|mm       | Error: Cannot find module 'module-alias/register'
0|mm       |     at Module._resolveFilename (internal/modules/cjs/loader.js:602:15)
0|mm       |     at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
0|mm       |     at Function.Module._load (internal/modules/cjs/loader.js:528:25)
0|mm       |     at Module.require (internal/modules/cjs/loader.js:658:17)
0|mm       |     at require (internal/modules/cjs/helpers.js:20:18)
0|mm       |     at Object.<anonymous> (/home/pi/MagicMirror/js/app.js:15:1)
0|mm       |     at Object.<anonymous> (/home/pi/MagicMirror/js/app.js:283:3)
0|mm       |     at Module._compile (internal/modules/cjs/loader.js:711:30)
0|mm       |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
0|mm       |     at Module.load (internal/modules/cjs/loader.js:620:32)

```

My configuration is
node --version v10.15.2
pm2 --version 4.2.1
npm --version 6.13.2
MM version 2.10.0
raspberry pi 3+ b

Any help will be much appreciated

MichMich, 2020-01-02:
Did you run `npm install` from within the MagicMirror folder?

sdetweil, 2020-01-02:
how did u upgrade?  git pull & npm install, or the upgrade script?   the upgrade script should have done the npm install for you

jsteel715, 2020-01-02:
I had the same problem after doing a git pull, forgot to run npm install. Worked fine after that :)

rsperezn, 2020-01-03:
Restarted the pi and it worked fine :man_shrugging:. Thanks

closed 2020-01-03 by rsperezn