MagicMirrorOrg/MagicMirror #1855: Update Notification stays on after updating

Okiev, 2020-01-02:
I did the update as described:

1.  bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/upgrade-script.sh)"
2.  	bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/upgrade-script.sh)" apply

After restart the Mirror it the update notification still appears, also after a reboot of the rpi. Am I'm doing something wrong? 

Thanks in advance!

sdetweil, 2020-01-02:
there is a bug in the master upgrade script (testing fix now)..  sorry, please use this one
see
https://forum.magicmirror.builders/topic/10859/new-update-upgrade-script-ready-for-testing

sdetweil, 2020-01-04:
yeh,   2832 commits..

I don't understand that either.. that is a git  message, not generated by MM code anywhere or the installer.. and is shown by the updatenotification  module.. direct from the github  master branch

if u do a git fetch/ git merge u get return code 1 , but it appears successful, but not complete
if you do SECOND git merge, you get 112 changes actually applied..

MichMich, 2020-01-08:
Can I close this issue?

Okiev, 2020-01-11:
Yes and thanks a lot!

closed 2020-01-11 by Okiev