MagicMirrorOrg/MagicMirror #1859: new run-start.sh fails detecting xwindows in docker setup

khassel, 2020-01-02:
**Platform**: Raspberry Pi 3, running magicmirror as docker image with `npm start` ([full setup](https://gitlab.com/khassel/magicmirror))
**Node Version**: 13.5.0
**MagicMirror Version**:  v2.10.0

**Description**: 
With the new release v2.10.0 the script run-start.sh was changed. The new version tries to handle some special cases falling back to `node serveronly` instead of starting electron.

- I think such a fallback does not lead to the desired results and make things more complicated. So if someone calls `npm start` he should know what he is doing and there should be no fallback to `node serveronly`. May you can create warnings but in the end the electron call should be executed.

- In a docker setup the following line (from run-start.sh) is not retrieving the expected result:

````
# check for xwindows running
xorg=$(pgrep Xorg)
````

`xorg` is empty because the Xorg-process is not running in the container but on the host and the container cant see this process. 

So calling `npm start` results here in starting `node serveronly` instead of electron and the mirror remains black.

**Possible Solutions**:
- my favorite: Remove the fallback to `node serveronly`, may display warnings.

- use `DISPLAY` variable setting your `xorg` variable (untestet):

remove the line `xorg=$(pgrep Xorg)` and use this to set `xorg`:

````
if [ -z "$DISPLAY" ]; then #If not set DISPLAY is SSH remote or tty
	export DISPLAY=:0 # Set by default display
else
	xorg=xwindowsrunning
fi
````

So Is there any chance to get a patched release fixing this bug?

Thanks,

Karsten.

MichMich, 2020-01-03:
This issue confirms the feeling I have had for a long time. I created a separate issue: https://github.com/MichMich/MagicMirror/issues/1860

Regarding your issue: feel free to send a PR.

sdetweil, 2020-01-03:
i don't understand this.. the browser is OUTSIDE the container, and so the docker container should be running in serveronly mode

we still have a number of new users running on armv6, and electron is no longer available on the latest node versions.   so, they cannot install (electron is optional now due to this)  and cannot run (npm start will fail as there is NO electron installed)..

sdetweil, 2020-01-03:
>So Is there any chance to get a patched release fixing this bug?

also, you can set serverOnly : false,  in config.js

altho I still don't see how this affects the container deployment. the browser is started outside the container, by some other script/mechanism, and should be unchanged

sdetweil, 2020-01-03:
also, the current docker container says on startup to do 

CMD ["node", "serveronly"]

which never executes the run-start.sh command

please point to the docker instance and setup you are running.. I don't think this is wrong, but I clearly misunderstand something

khassel, 2020-01-03:
> Regarding your issue: feel free to send a PR.

done: https://github.com/MichMich/MagicMirror/pull/1861

khassel, 2020-01-03:
> anyhow, change the config.js to serveronly: false
> and u will be ok

this can I do only for my personal config.js, but not for all users which use their own config.js with this docker image ...
 
> or maybe easier for your custom setup is replace run-start.sh with your own with
> electron js/electron.js $1;

I did this yesterday, but I hate such ugly patches. And I think this should be fixed here.

> we still have a number of new users running on armv6, and electron is no longer available on the latest node versions. so, they cannot install (electron is optional now due to this) and cannot run (npm start will fail as there is NO electron installed)..

The idea is good to enable such users to run magicmirror. 
But why not provide a standalone-script for this usecase, instead of trying to handle "all" cases in `npm start`?
You will never know all side effects ...

MichMich, 2020-01-03:
> But why not provide a standalone-script for this usecase, instead of trying to handle "all" cases in npm start?

I agree with this. I think it is fair to give the user some ‘responsibility’ as well. This aligns with issue #1860.

sdetweil, 2020-01-03:
Why does the rocket scientist user get a break, but the newbies don't?  I haven't had more than 5 docker user requests, but gave new user failures every day. Node versions, npm version, electron versions, os version aging. And all the modules subject to all this.

I don't know how u find a simple solution anymore. 

Pi0 users can't get a current version of electron, on the versions of node available now. So we had to make it optional so mm would at least finish npm install. 

But now it can't start. Why should they be penalized?
Buster guys need the newer electron, but test system is broken. Why is it the users fault. 

The pull/npm install doesn't work, except for a tiny portion of the users . why do the users have to suffer because of an update ( and another one in 90 days).

MichMich, 2020-01-04:
@sdetweil I think this should be part of the #1860 discussion. But just to be clear. Nobody should be suffering. The only thing I'm proposing is to operate the installer from the core and make two repo's. "rocket scientists" kan use the core without the installer. All others can get the same results by using the separate installer. But as said: let's continue that discussion in #1860.

MichMich, 2020-01-08:
Fix merged.

PR #1861 (2020-01-08): fix run-start.sh: if running in docker-container, just start electron…
Fixes #1859

With this fix the `run-start.sh` script differentiate if running in a docker-container or not:

-  running inside a docker-container: Just start electron
-  running not in a docker-container: Do things as before
closed 2020-01-08 by MichMich