MagicMirrorOrg/MagicMirror #1862: After last update MM2 won't start

paaland, 2020-01-03:
**Platform**:  
Raspberry Pi 3

**Node Version**: 
v10.18.0

**MagicMirror Version**:  
JavaScript version (v2).

**Description**: 
Got "new version available" message so I ssh'ed in and did a git pull. Now it now longer wants to start.

**Steps to Reproduce**: 

**Expected Results**: 
Working MM2.

**Actual Results**: 
Get log files lists this error:

App threw an error during load
Error: Cannot find module 'module-alias/register'
    at Module._resolveFilename (internal/modules/cjs/loader.js:602:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (internal/modules/cjs/loader.js:528:25)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/js/app.js:15:1)
    at Object.<anonymous> (/home/pi/MagicMirror/js/app.js:283:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
A JavaScript error occurred in the main process
Uncaught Exception:
Error: Cannot find module 'module-alias/register'
    at Module._resolveFilename (internal/modules/cjs/loader.js:602:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (internal/modules/cjs/loader.js:528:25)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/js/app.js:15:1)
    at Object.<anonymous> (/home/pi/MagicMirror/js/app.js:283:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)

MichMich, 2020-01-03:
You forgot to run `npm install` as mentioned in the release notes. That should solve it.

closed 2020-01-03 by MichMich

paaland, 2020-01-03:
npm install fixed it....