MagicMirrorOrg/MagicMirror #1864: config.js file Calendar part, don't show the 4th ICS and mess up the date format

tpbillund, 2020-01-04:
Hi
I have two issues with the new 2.10 of MM.
1. have added a 4th. ICS
(https://calendar.google.com/calendar/ical/da.danish%23holiday%40group.v.calendar.google.com/public/basic.ics) and I can't get it to show the dates in the "calendars" the 3 other ICS is working fine
`
calendars: [
          {
                symbol: "calendar",
                url: "1. LINK TO WORKING GOOGLE ICS"
          },
          {
                symbol: "birthday-cake",
                url: "2. LINK TO WORKING GOOGLE ICS"
          },
          {
                symbol: "calendar",
                url: "3. LINK TO WORKING GOOGLE ICS"
          },
         {
                symbol: "calendar",
                url: "https://calendar.google.com/calendar/ical/da.danish%23holiday%40group.v.calendar.google.com/public/basic.ics"
          }
]
`

2. Im using fullDayEventDateFormat: "DD MMM" but appears as "Do MMM" (21. jan - maybe) for every second lines and "MMM Do" (Jan 25. - maybe) for the others in between.
`
config: {
           fullDayEventDateFormat: "DD MMM",
           timeFormat: "abdolute",
           urgency: 7,
           calendars: [
                    ..........
]
}
`
Can someone help with...
- Why the 4th ICS link is not showed
- Why i get the two datetypes Do MMM and MMM Do, when I im my config.js file have DD MMM

sdetweil, 2020-01-04:
timeFormat: "abdolute",

should be "absolute  "    s not d

sdetweil, 2020-01-04:
on the cal entries..looking at the calendar, the 1st events in 2020 are in february
from the file, start dates and lines listed
```
Line 372	 DTSTART;VALUE=DATE	20190101
Line 36	   DTSTART;VALUE=DATE	20190214
Line 666	 DTSTART;VALUE=DATE	20190303
Line 120	 DTSTART;VALUE=DATE	20190331
Line 330	 DTSTART;VALUE=DATE	20190414
Line 204	 DTSTART;VALUE=DATE	20190418
Line 456	 DTSTART;VALUE=DATE	20190419
Line 246	 DTSTART;VALUE=DATE	20190421
Line 792	 DTSTART;VALUE=DATE	20190422
Line 750	 DTSTART;VALUE=DATE	20190501
Line 708	 DTSTART;VALUE=DATE	20190505
Line 78	   DTSTART;VALUE=DATE	20190517
Line 498	 DTSTART;VALUE=DATE	20190530
Line 624	 DTSTART;VALUE=DATE	20190605
Line 288	 DTSTART;VALUE=DATE	20190609
Line 834	 DTSTART;VALUE=DATE	20190610
Line 162	 DTSTART;VALUE=DATE	20191027
Line 582	 DTSTART;VALUE=DATE	20191224
Line 540	 DTSTART;VALUE=DATE	20191225
Line 876	 DTSTART;VALUE=DATE	20191226
Line 414	 DTSTART;VALUE=DATE	20191231
Line 358	 DTSTART;VALUE=DATE	20200101
above in the past

Line 22	   DTSTART;VALUE=DATE	20200214
Line 652	 DTSTART;VALUE=DATE	20200223
Line 106	 DTSTART;VALUE=DATE	20200329
Line 316	 DTSTART;VALUE=DATE	20200405
Line 190	 DTSTART;VALUE=DATE	20200409
Line 442	 DTSTART;VALUE=DATE	20200410
Line 232	 DTSTART;VALUE=DATE	20200412
Line 778	 DTSTART;VALUE=DATE	20200413
Line 736	 DTSTART;VALUE=DATE	20200501
Line 694	 DTSTART;VALUE=DATE	20200505
Line 64	   DTSTART;VALUE=DATE	20200508
Line 484	 DTSTART;VALUE=DATE	20200521
Line 274	 DTSTART;VALUE=DATE	20200531
Line 820	 DTSTART;VALUE=DATE	20200601
Line 610	 DTSTART;VALUE=DATE	20200605
Line 148	 DTSTART;VALUE=DATE	20201025
Line 568	 DTSTART;VALUE=DATE	20201224
Line 526	 DTSTART;VALUE=DATE	20201225
Line 862	 DTSTART;VALUE=DATE	20201226
Line 400	 DTSTART;VALUE=DATE	20201231

next year 
Line 344	 DTSTART;VALUE=DATE	20210101
Line 8	   DTSTART;VALUE=DATE	20210214
Line 638	 DTSTART;VALUE=DATE	20210214
Line 92	   DTSTART;VALUE=DATE	20210328
Line 302	 DTSTART;VALUE=DATE	20210328
Line 176	 DTSTART;VALUE=DATE	20210401
Line 428	 DTSTART;VALUE=DATE	20210402
Line 218	 DTSTART;VALUE=DATE	20210404
Line 764	 DTSTART;VALUE=DATE	20210405
Line 50	   DTSTART;VALUE=DATE	20210430
Line 722	 DTSTART;VALUE=DATE	20210501
Line 680	 DTSTART;VALUE=DATE	20210505
Line 470	 DTSTART;VALUE=DATE	20210513
Line 260	 DTSTART;VALUE=DATE	20210523
Line 806	 DTSTART;VALUE=DATE	20210524
Line 596	 DTSTART;VALUE=DATE	20210605
Line 134	 DTSTART;VALUE=DATE	20211031
Line 554	 DTSTART;VALUE=DATE	20211224
Line 512	 DTSTART;VALUE=DATE	20211225
Line 848	 DTSTART;VALUE=DATE	20211226
Line 386	 DTSTART;VALUE=DATE	20211231
```

tpbillund, 2020-01-05:
> 
> 
> timeFormat: "abdolute",
> 
> should be "absolute " s not d

Yes sry it is with S in my code, was hitting D in here, sry.
Have added screenshots of code

tpbillund, 2020-01-05:
> on the cal entries..looking at the calendar, the 1st events in 2020 are in february
> from the file, start dates and lines listed
>
> ```
> Line 372	 DTSTART;VALUE=DATE	20190101
> Line 36 [...]

Correct :-) I can see that to in the file/link, but I get nothing in my Magic Mirror and i have privats event from other ical showed, thats in 90 days.

closed 2020-03-12 by stale[bot]