MagicMirrorOrg/MagicMirror #1866: Install script hangs on some systems doing apt-get upgrade

sdetweil, 2020-01-06:
updated install script prompts for y/n to continue on some systems. 

stdout redirected to file, so user can't see it.

LeviDavisCode, 2020-01-06:
hey did you have any luck getting it to work? I have the same problem

sdetweil, 2020-01-06:
@LeviDavisCode 
yes, use the updated script, see the external script repo for the new url

https://github.com/sdetweil/MagicMirror_scripts

vkramsingh, 2020-01-07:
Hi,
I used the external script url but my installation is still stuck at apt-update. The last line in the log file is apt-get update completed ok

sdetweil, 2020-01-07:
@vkramsingh 
please use the install script from  here
https://github.com/sdetweil/MagicMirror_scripts

MichMich, 2020-01-10:
FYI, The link to (broken) the install script is removed from the README.md.

tbouron, 2020-01-12:
Experienced the same issue when I reinstalled my mirror this weekend. It's because the script uses `sudo` to run the majority of commands. But for some reason, the script does not prompt for the password, hence the hang.

Once you run at least once command with `sudo` before hand, the password is cached and the script will work fine.

I don't know if this is a regression or not, last time it worked fine but I might have used `sudo` prior to executing the script.

sdetweil, 2020-01-12:
the newer updated script does handle this..   the new doc points to the alternate install sites

sdetweil, 2020-01-12:
I am closing this, link to new site 
https://github.com/sdetweil/MagicMirror_scripts

closed 2020-01-12 by sdetweil