MagicMirrorOrg/MagicMirror #1867: SyntaxError: Unexpected token ...

ThePirateWhoSmellsOfSunflowers, 2020-01-06:
Hi!
After the last update b595cdd31e821fc291e852cb58c17c6e4c9089c5, my MagicMirror install no longer works. I'm using the master branch on a Raspberry Pi 2. I only use MM in `serveronly` mode.

The stack trace is:
```
xxx@yyy:~/MagicMirror $ git branch
* master
xxx@yyy:~/MagicMirror $ git pull
Already up-to-date.
xxx@yyy:~/MagicMirror $ node -v
v6.11.2
xxx@yyy:~/MagicMirror $ node serveronly/
/home/xxx/MagicMirror/node_modules/ws/lib/websocket.js:347
      ...options
      ^^^

SyntaxError: Unexpected token ...
    at createScript (vm.js:56:10)
    at Object.runInThisContext (vm.js:97:10)
    at Module._compile (module.js:542:28)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/xxx/MagicMirror/node_modules/ws/index.js:3:19)
```
(`npm install` is omitted here but I did it)

Thanks.

:sunflower:

MichMich, 2020-01-06:
Try upgrading node. V6 is pretty old.

ThePirateWhoSmellsOfSunflowers, 2020-01-06:
Thanks, upgrading node fixes the issue.
For information (and if someone in the future stumble upon this issue), the last node version compiled for raspberry pi 2 (armv6) is [11.15.0](https://nodejs.org/dist/v11.15.0/node-v11.15.0-linux-armv6l.tar.gz).

:sunflower:

closed 2020-01-06 by ThePirateWhoSmellsOfSunflowers