MagicMirrorOrg/MagicMirror #1868: Images resized

gielekee, 2020-01-07:
Yesterday I updated MagicMirror. After the update images started resizing out of nowhere.

I already reinstalled the modules that we're effected by this.

I hope somebody has a solution for this because on the internet I couldn't find anything.

MichMich, 2020-01-07:
What images do you mean?

gielekee, 2020-01-07:
This image with the MMM-NowPlayingOnSpotify module.
With other images i have the same thing
[image]

MichMich, 2020-01-07:
This is a CSS issue in the `MMM-NowPlayingOnSpotify` app. Please open a issue in the respective model repo.

sdetweil, 2020-01-07:
@gielekee   how did you update?  and from what version? 

this sounds like the old custom.css overlayed problem, before 2.9

closed 2020-01-08 by MichMich