MagicMirrorOrg/MagicMirror #1879: MaxListenersExceededWarning

sergge1, 2020-01-16:
I am running MM v2.10.1
Today I have a look at terminal and have found an ERROR message (the full log show below). Error appeared just 50 minutes after the MM start (once per each start).

> [23:40:52.148] [ERROR]  (node:13645) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 message listeners added. Use emitter.setMaxListeners() to increase limit
 
MM still works despite this error but I am wondering what is this ERROR about and how to manage it.

**Full terminal log**
```
**pi@raspberrypi:~/MagicMirror $ sudo npm start**
magicmirror@2.10.1 start /home/pi/MagicMirror
./run-start.sh
[22:50:42.019] [LOG]    Starting MagicMirror: v2.10.1
[22:50:42.036] [LOG]    Loading config ...
[22:50:42.059] [LOG]    Loading module helpers ...
[22:50:42.062] [LOG]    No helper found for module: alert.
[22:50:42.064] [LOG]    No helper found for module: clock.
[22:50:43.281] [LOG]    Initializing new module helper ...
[22:50:43.283] [LOG]    Module helper loaded: calendar
[22:50:43.287] [LOG]    No helper found for module: currentweather.
[22:50:43.289] [LOG]    No helper found for module: weatherforecast.
[22:50:43.430] [LOG]    Initializing new module helper ...
[22:50:43.432] [LOG]    Module helper loaded: newsfeed
[22:50:43.561] [LOG]    Initializing new module helper ...
[22:50:43.562] [LOG]    Module helper loaded: MMM-PIR-Sensor
[22:50:43.566] [LOG]    No helper found for module: MMM-SmartWebDisplay.
[22:50:44.144] [LOG]    Initializing new module helper ...
[22:50:44.145] [LOG]    Module helper loaded: MMM-mqtt
[22:50:44.360] [LOG]    Initializing new module helper ...
[22:50:44.361] [LOG]    Module helper loaded: MMM-Remote-Control
[22:50:44.362] [LOG]    All module helpers loaded.
[22:50:44.363] [LOG]    Starting server on port 8181 ... 
[22:50:44.376] [INFO]   You're using a full whitelist configuration to allow for all IPs
[22:50:44.393] [LOG]    Server started ...
[22:50:44.396] [LOG]    Connecting socket for: calendar
[22:50:44.397] [LOG]    Starting node helper for: calendar
[22:50:44.399] [LOG]    Connecting socket for: newsfeed
[22:50:44.401] [LOG]    Starting module: newsfeed
[22:50:44.402] [LOG]    Connecting socket for: MMM-PIR-Sensor
[22:50:44.403] [LOG]    Connecting socket for: MMM-mqtt
[22:50:44.405] [LOG]    MMM-mqtt started ...
[22:50:44.406] [LOG]    Connecting socket for: MMM-Remote-Control
[22:50:44.408] [LOG]    Starting node helper for: MMM-Remote-Control
[22:50:44.436] [LOG]    Sockets connected & modules started ...
[22:50:45.046] [LOG]    Launching application.
[22:50:52.218] [LOG]    Create new calendar fetcher for url: https://calendar.google.com/calendar......./basic.ics - Interval: 300000
[22:50:52.242] [LOG]    Create new calendar fetcher for url: https://calendar.google.com/calendar......./basic.ics - Interval: 300000
[22:50:52.248] [LOG]    Create new news fetcher for url: https://www..../rss/ - Interval: 300000
[22:50:52.367] [LOG]    Creating new MQTT client for url:  mqtt://localhost:1883
**[23:40:52.148] [ERROR]  (node:13645) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 message listeners added. Use emitter.setMaxListeners() to increase limit**
```
**I am running MM on the following hardware/software:**

>Raspbian version
```
pi@raspberrypi:~/MagicMirror $ cat /etc/os-release
>PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
>NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
```

>Linux core
```
pi@raspberrypi:~/MagicMirror $ uname -a
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
```
>Hardware model
```
pi@raspberrypi:~/MagicMirror $ cat /proc/device-tree/model
Raspberry Pi 3 Model B Plus Rev 1.3
```
>Node and NPM version

`pi@raspberrypi:~ $ node -v
v10.18.1`
`pi@raspberrypi:~ $ npm -v
6.13.4`

List of active MM modules
```- alert
- clock
- calendar
- currentweather
- weatherforecast
- newsfeed
- MMM-PIR-Sensor
- MMM-SmartWebDisplay
- MMM-mqtt
- MMM-Remote-Control
```
Full version of [config.js](https://github.com/sergge1/bugs/blob/master/config.js) is here

MichMich, 2020-01-16:
Could you please test your Mirror without the 3d-party modules? Thanks!

sergge1, 2020-01-16:
will try right now. results will be in 50 min

sergge1, 2020-01-16:
works without errors while 3d-party modules are disabled.
Will allow them one-by-one to check which one make this error and will report here.

MichMich, 2020-01-16:
If one specific module gives an error, please create an issue in the module's respective repository. If the developer of that module finds the issue is caused by the core, the developer can create an issue in this repo.