MagicMirrorOrg/MagicMirror #1890: Feature request: Support config override via query parameters

adi-miller, 2020-01-19:
I'm using MM (which is AWESOME btw!) on two screens at home. One is the raspi running server+client, and the other is on an old tablet which I configured to open in kiosk mode and load http://mm2:8080 in fullscreen. This usually works great. 

Some of the modules are less relevant on the tablet screen, and I would like to hide them. A simple solution I thought about was to support in MM the ability to override config parameters via query parameters, so that for example, on my tablet I would call: 

http://mm2:8080?currentweather.disable=true

I realize that this can't work for all modules, since some use such parameters in the server-side logic, which is running once, but I believe this can still be useful for some client-side logic. 

Thoughts?

MichMich, 2020-01-19:
Best approach is to write your own module that controls other modules. This is the perfect case for a custom module.

adi-miller, 2020-01-19:
I'm not sure how to address the question of providing the screen context to the module. In other words, how would the module know which screen it is "running on"?

MichMich, 2020-01-19:
It probably runs on a different resolution so you could check that. Or check the IP adres. Or check the browser agent. ... there are many ways to accomplish this. :)

buxxi, 2020-01-20:
You could probably use a custom.css with [media queries](https://developer.mozilla.org/en-US/docs/Web/CSS/Media_Queries/Using_media_queries) and just set those module classes to `display: none`, would probably be easier than creating a new module.

closed 2020-03-27 by stale[bot]