MagicMirrorOrg/MagicMirror #1891: TLS 1.3 Support for MM and Calendar

Pakmann83, 2020-01-20:
**Platform**: Raspberry PI 3 - Buster

**Node Version**: Standard on MM 2.9.0

**MagicMirror Version**: MM 2.9.0, PHP 7.3

**Description**: 
a few weeks ago i had a chat with Sam regarding TLS1.3 Support. We Need newer Libs for NodeJs (v12) etc. This was no problem to reinstall MM with newer Packages. Problem was the Calendar Module which Not Support TLS1.3.

A complete solution should be usefull to be state of the art

See: https://forum.magicmirror.builders/topic/12051/tls-1-3-support-for-mm-and-calendar

**Steps to Reproduce**: see on the top

**Expected Results**: Run MM and all nessary moduls with TLS1.3 support - Especially Calendar modul with an connection to external calendars with are configured for TLS 1.3

**Actual Results**: Only TLS 1.2 is supported

**Configuration**: not relevant here

**Additional Notes**:

Pakmann83, 2020-03-24:
I Wang Kindly ask in It is possible to have an solution for it?

khassel, 2020-03-24:
may you can provide an example calendar url with TLS1.3?

Pakmann83, 2020-03-29:
@khassel i am sorry. I Tried it with my own cloud/ calender. I am sure that other provider already used TLS1.3

closed 2020-06-04 by stale[bot]