MagicMirrorOrg/MagicMirror #1907: updatenotification creates zombie processes

buxxi, 2020-01-27:
I noticed my server had lots of zombie-processes where the parent was the container running magicmirror in serveronly-mode. By enabling only one module at a time I managed to locate the problematic module which is the updatenotification.

The cause of this seems to be partly my fault due to cloning my own modules with ssh instead of https (using url = git@github.com:xxx/xxx.git) and I guess this fails in the container since it doesn't have access to my username/password/ssh-keys.

But it's probably also something missing in the code that doesn't close the connection after the update-check, since I've seen a few git-http zombies too.

Would also be great to have a blacklist for modules not to update-check or at least a flag to only update check for the main project.

MichMich, 2020-01-31:
I agree. Feel free to send a PR. I'll leave this open so one of us can investigate this later.

buxxi, 2020-02-09:
Should it be in the configuration for the updatenotification-module or as part of each module?
```
{
	module: "updatenotification",
	config: {
		moduleBlacklist: ["clock"] //empty array is default
	}
}
```
vs
```
{
	module: "clock",
	updateCheck: false //default is true
}
```

MichMich, 2020-02-09:
I prefer the first one. Thanks!

MichMich, 2020-02-09:
And maybe call it `ignoreModules: []`?

buxxi, 2020-02-11:
Made a PR for the ignoreModule functionality, guess I should make a PR for the documentation in the other repo now that its separate?.

This is the line that creates the zombie processes, the callback function is never called at all (I guess a timeout needs to be specified in some way, but can't really find it):
https://github.com/MichMich/MagicMirror/blob/d7295948fdb322c3bb267c08fb634719aa3f9fa5/modules/default/updatenotification/node_helper.js#L70

Found 2 other smaller problems when trying to debug this:
- https://github.com/MichMich/MagicMirror/blob/d7295948fdb322c3bb267c08fb634719aa3f9fa5/modules/default/updatenotification/node_helper.js#L41 this line isn't blocking, so its only the "default"-module that will be checked the first time. Should probably convert this code to use Promises instead.
- The `refreshInterval` configuration isn't in the documentation

buxxi, 2020-02-18:
Have done a bit more research about this.

As stated above the fetch()-method is never completed with error nor success.
It seems to be stuck waiting for keyboard-input from the user for this:
```
The authenticity of host '<host> (<ip>)' can't be established.
RSA key fingerprint is <fingerprint>.
Are you sure you want to continue connecting (yes/no)?
```

And when its run the next time it creates another process with the same issue. I tried to accept the host manually and then it goes to the error-handler for fetch() since it can't fetch it without my ssh-keys.

Adding this seems to make it not hang even without accepting the host:
```
	sg.git.env("GIT_SSH_COMMAND", "ssh -o BatchMode=yes");
```

Not really sure if that is a proper solution or if remotes that isn't http/https just should not be checked.

khassel, 2020-06-05:
> remotes that isn't http/https just should not be checked.

I would prefer this solution.

My setup seems similar to yours (docker container, module with `git@...` as remote) but I have no such zombie-processes, but errors in the container logs (which is o.k. for me):
````
[2020-06-05 23:00:16.174] [LOG]    Starting node helper for: mmm-remote-docker
[2020-06-05 23:00:16.175] [LOG]    Connecting socket for: newsfeed
[2020-06-05 23:00:16.175] [LOG]    Starting module: newsfeed
[2020-06-05 23:00:16.176] [LOG]    Sockets connected & modules started ...
[2020-06-05 23:00:16.176] [LOG]
Ready to go! Please point your browser to: http://0.0.0.0:8080
[2020-06-05 23:00:16.823] [ERROR]  Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
````

simphide, 2020-06-21:
this problem is still present. I've never cloned a module using ssh - only http.

Same setup with docker server-only...

simphide, 2020-06-27:
I noticed today, that the zombie processes occurr when the magicmirror has no internet

MichMich, 2020-07-04:
Maybe we should check to see if internet is working correctly before we fetch the repos? A PR on this is welcome!

buxxi, 2020-10-01:
Seems more like a workaround than a fix, what if the call for internet connectivity works but the update check call fails?

Proper timeout-handling and closing the process after the call is done is probably the correct way. Which I can't find in the documentation for simple-git, but there must be some way to do it? 

Maybe update the simple-git dependency, there has been many new releases since the one specified in package.json?

FrankBlackMG, 2020-10-06:
A while back I reviewed all of the changes from the current version that MM uses and the current simple-git version and I didn't see any breaking changes. I also pulled in the latest on my local (2.20.1) and didn't see any issues.

buxxi, 2021-02-06:
Still happens in the current release, the update of simple-git didn't solve it.
```
> ps ax | grep "Z"
 430199 ?        Z      0:00 [git-remote-http] <defunct>

> pstree -p -s 430199
systemd(1)───containerd(385398)───containerd-shim(411902)───node(411920)───git-remote-http(430199)
```

And the only thing running in a container on the server and using node is magicmirror.

MichMich, 2021-02-06:
Maybe we should open een issue on the simple-git repo? Or is this a MagicMirror specific issue?

rejas, 2021-02-06:
Mhhh... So can we assume that it is an issue in the simple-git library? Maybe open a bug report there and see what they say?

rejas, 2021-02-06:
Damn @MichMich beat me by a second :-D

MichMich, 2021-02-06:
Muhahaha

buxxi, 2021-03-06:
Since my bug report the dev of simple-git (git-js) has added the possibility to set timeout for the git commands (see https://github.com/steveukx/git-js/blob/main/docs/PLUGIN-TIMEOUT.md) in v.2.36.

Is there anything stopping us from using async/await in the node_helper?

rejas, 2021-03-06:
> Is there anything stopping us from using async/await in the node_helper?

Only the question if we loose support for some browsers: https://caniuse.com/async-functions 
ios10 is the only questionable here though I think

buxxi, 2021-03-06:
But node_helper isn't run on the client?

rejas, 2021-03-06:
Then there is no reason for me to object :-) So, you want to go for a PR then?

buxxi, 2021-03-13:
Hopefully the PR I made will solve this, kinda hard to know without running the application for weeks.

buxxi, 2021-04-16:
Bah, been running the new version since release and logged in to my mirror today and was greeted with this:
`=> There are 13 zombie processes.`

And all of those are still like this:
`11487 ?        Z      0:00 [git-remote-http] <defunct>`

Annoying...

rejas, 2021-04-16:
Annyoing indeed. How does one get that output to check on his own mirror?

buxxi, 2021-04-17:
The first output is from MOTD when I log in to my Ubuntu server, so I don't really know how to reproduce that on a RPi.

But all zombie processes can easily be found by: 
```console
ps ax | grep Z
  11487 ?        Z      0:00 [git-remote-http] <defunct>
```
To see where the process belongs to you can make this command:
```console
pstree -s 11487
systemd───containerd───containerd-shim───node───git-remote-http
```
To see that it's node that spawning the process that becomes a zombie, the node process belongs to a container which is started by systemd in my case.

khassel, 2021-04-17:
I can now confirm these zombie processes.

In the logs I found:
````
[16.04.2021 07:26.44.533] [ERROR] Failed to fetch git data for MMM-RepoStats: Error: block timeout reached
[16.04.2021 14:07.37.595] [ERROR] Failed to fetch git data for MMM-RepoStats: Error: block timeout reached
[16.04.2021 18:18.12.722] [ERROR] Failed to fetch git data for MMM-RMV: Error: block timeout reached
[17.04.2021 03:29.23.095] [ERROR] Failed to fetch git data for MMM-RMV: Error: block timeout reached
[17.04.2021 03:29.24.467] [ERROR] Failed to fetch git data for MMM-Strava: Error: block timeout reached
[17.04.2021 09:30.11.592] [ERROR] Failed to fetch git data for MMM-RMV: Error: block timeout reached
[17.04.2021 09:30.13.377] [ERROR] Failed to fetch git data for default: Error: block timeout reached
[17.04.2021 16:01.01.967] [ERROR] Failed to fetch git data for default: Error: block timeout reached
````

I had 8 zombie processes so may this is a 1:1 relation to the above logs.

The message `block timeout reached` is coming from [this line](https://github.com/steveukx/git-js/blob/256acf3179f916c8a369492de6dea44691d03f63/test/integration/timeout-plugin.spec.ts#L22).

khassel, 2021-04-17:
I could reproduce this error for the mm-repo with every restart.

This is may caused by the default timout=1000 which seems to low.

I'm running now with timeout=15000 and it works (the error after start is gone), will see if it occours later.

One explanation for this is may that I´m using my docker setup where the mm-git-repo is cloned with `git clone --depth 1 -b "${BuildRef}" --single-branch "${GitRepo}"` so performing here a `git fetch --dry-run` will bring a long list of missing refs:

````
/opt/magic_mirror $ git fetch --dry-run
From https://github.com/MichMich/MagicMirror
   fd1913a7..ccf5bb93  develop    -> origin/develop
 * [new tag]           V2.2.1     -> V2.2.1
 * [new tag]           v.2.1.3    -> v.2.1.3
 * [new tag]           v2.0.0     -> v2.0.0
 * [new tag]           v2.0.1     -> v2.0.1
 * [new tag]           v2.0.2     -> v2.0.2
 * [new tag]           v2.0.3     -> v2.0.3
 * [new tag]           v2.0.4     -> v2.0.4
 * [new tag]           v2.0.5     -> v2.0.5
 * [new tag]           v2.1.0     -> v2.1.0
 * [new tag]           v2.1.1     -> v2.1.1
 * [new tag]           v2.1.2     -> v2.1.2
 * [new tag]           v2.10.0    -> v2.10.0
 * [new tag]           v2.10.1    -> v2.10.1
 * [new tag]           v2.11.0    -> v2.11.0
 * [new tag]           v2.12.0    -> v2.12.0
 * [new tag]           v2.13.0    -> v2.13.0
 * [new tag]           v2.14.0    -> v2.14.0
 * [new tag]           v2.15.0    -> v2.15.0
 * [new tag]           v2.2.0     -> v2.2.0
 * [new tag]           v2.2.2     -> v2.2.2
 * [new tag]           v2.3.0     -> v2.3.0
 * [new tag]           v2.3.1     -> v2.3.1
 * [new tag]           v2.4.0     -> v2.4.0
 * [new tag]           v2.4.1     -> v2.4.1
 * [new tag]           v2.5.0     -> v2.5.0
 * [new tag]           v2.6.0     -> v2.6.0
 * [new tag]           v2.7.0     -> v2.7.0
 * [new tag]           v2.7.1     -> v2.7.1
 * [new tag]           v2.8.0     -> v2.8.0
 * [new tag]           v2.9.0     -> v2.9.0
 * [new tag]           V2.2.1     -> V2.2.1
 * [new tag]           v.2.1.3    -> v.2.1.3
 * [new tag]           v2.0.0     -> v2.0.0
 * [new tag]           v2.0.1     -> v2.0.1
 * [new tag]           v2.0.2     -> v2.0.2
 * [new tag]           v2.0.3     -> v2.0.3
 * [new tag]           v2.0.4     -> v2.0.4
 * [new tag]           v2.0.5     -> v2.0.5
 * [new tag]           v2.1.0     -> v2.1.0
 * [new tag]           v2.1.1     -> v2.1.1
 * [new tag]           v2.1.2     -> v2.1.2
 * [new tag]           v2.10.0    -> v2.10.0
 * [new tag]           v2.10.1    -> v2.10.1
 * [new tag]           v2.11.0    -> v2.11.0
 * [new tag]           v2.12.0    -> v2.12.0
 * [new tag]           v2.13.0    -> v2.13.0
 * [new tag]           v2.14.0    -> v2.14.0
 * [new tag]           v2.15.0    -> v2.15.0
 * [new tag]           v2.2.0     -> v2.2.0
 * [new tag]           v2.2.2     -> v2.2.2
 * [new tag]           v2.3.0     -> v2.3.0
 * [new tag]           v2.3.1     -> v2.3.1
 * [new tag]           v2.4.0     -> v2.4.0
 * [new tag]           v2.4.1     -> v2.4.1
 * [new tag]           v2.5.0     -> v2.5.0
 * [new tag]           v2.6.0     -> v2.6.0
 * [new tag]           v2.7.0     -> v2.7.0
 * [new tag]           v2.7.1     -> v2.7.1
 * [new tag]           v2.8.0     -> v2.8.0
 * [new tag]           v2.9.0     -> v2.9.0
````

Maybe this exceeds the default timeout.

But this is not an explanation for other modules where this list is very short.

khassel, 2021-04-20:
After increasing the timeout the zombie processes are gone.

@buxxi can you test this too?

If you can confirm this we should increase the default timeout [here](https://github.com/MichMich/MagicMirror/blob/develop/modules/default/updatenotification/updatenotification.js#L13).

blackthroat, 2021-04-22:
@khassel I recently upgraded to MagicMirror: v2.15.0 and was digging in the log files as a 3rd party module was preventing it from loading. I noticed that I was also receiving `Error: block timeout reached` for each of my modules on startup. After adjusting the timeout from `1000` to `15000`, it no longer threw that error.

buxxi, 2021-04-24:
Well, increasing the timeout just makes it less likely to happen (and sure 1000 was a bit low as default). I had this issue a few times when there was no timeout at all.
This just means the timeout-handling isn't working as it should and the process probably needs to be killed since it can't quit by itself.

sdetweil, 2021-04-24:
and as a side note, I have a private repo that gets the same error..

khassel, 2021-04-25:
We have 2 problems here:

- the default timeout is to low which produces `Error: block timeout reached` and the updatenotification for the affected module does not work. My tests with 5000 (instead of 1000) are working
- running MM as docker container produces additional a zombie process for every above error

I will change my docker setup to get rid of this error. Meanwhile you can add the `--init` parameter to your `docker run ...` command or `init: true` in the `docker-compose.yml` file. You can read more about this error running `npm` in a docker container e.g. [here](https://maximorlov.com/process-signals-inside-docker-containers/).

khassel, 2021-04-28:
@buxxi can you please test again and finally close this issue if it works on your side?

buxxi, 2021-05-01:
Ok, updated my container and lowered the timeout to 200ms, will check back in a day or two if the zombies are gone.

buxxi, 2021-05-06:
4 errors in the log and no zombies, so I guess we can call it fixed now. Thanks @khassel!

closed 2021-05-06 by buxxi

geoffrow, 2021-08-01:
Hi Guys, I have all the same problems. I fixed it... sort of. I went into the updatenotification.js and changed the time to a ridiculous number 

.register("updatenotification", {
        // Define module defaults
        defaults: {
                updateInterval: 10 * 60 * 100000000, // A VERY LONG TIME 
                refreshInterval: 24 * 60 * 60 * 1000, // one day
                ignoreModules: [],
                timeout: 5000
it came back with an error when I rebooted.
I went back and modified the update interval to something more reasonable. It now works
.register("updatenotification", {
        // Define module defaults
        defaults: {
                updateInterval: 10 * 60 * 10000, // every 10 minutes
                refreshInterval: 24 * 60 * 60 * 1000, // one day
                ignoreModules: [],
                timeout: 5000
HOPE THIS HELPS SOMEONE

PR #2479 (2021-03-14): Timeout handling for updatenotification
Changes that hopefully fixes #1907.

- Refactored the module to use async/await to make the error handling easier.
- Upgraded simple-git to use the timeout settings.
- Added timeout as config-property to the module with the default of 10 seconds (per module check)
- Changed the notification parameter to be its own object and not piggybacking on the objects returned from simple-git.

I have tested it by:
- hard resetting a modules git to a previous version
- dropping packages with iptables to the target address to force a timeout