MagicMirrorOrg/MagicMirror #1913: Electron Doesn't opening

vishwapinnawala, 2020-01-31:
[image]

works fine with chrome when i go to the link but it doesn't open automatically in electron Anyone had this issue ?

MichMich, 2020-01-31:
Could you give the following command a try from within the MagicMirror folder?
```
DISPLAY=:0 ./node_modules/.bin/electron js/electron.js
```

khassel, 2020-01-31:
may related to #1859 
Looking at his logs the run-start.sh starts `node serveronly` instead of electron.

MichMich, 2020-01-31:
Working on removing the ‘run-start.sh’. That’s why I want to test the mentioned line.

vishwapinnawala, 2020-02-01:
hell yeah it works!!!

vishwapinnawala, 2020-02-01:
added this as khassel says in his thread 

if [ -z "$DISPLAY" ]; then #If not set DISPLAY is SSH remote or tty
	export DISPLAY=:0 # Set by default display
else
	xorg=xwindowsrunning
fi

no it works normally

MichMich, 2020-02-01:
Could you please try if the `develop` branch solves your issue? Run the following commands from within your MagicMirror folders:
```
git checkout develop

git pull

npm install

npm start
```

vishwapinnawala, 2020-02-03:
[image]

khassel, 2020-02-07:
can you try `git checkout -b develop --track origin/develop` instead?

In case of error, can you provide more info:
* `git branch -a`
* `git show-ref`
* `git version`

sdetweil, 2020-02-10:
@vishwapinnawala 

can u tell me your environment? 

run-start tries to enable running in more env than the raw start..

if u could add -x to the bash line
and start and post the results of the npm start
so I can see the reason for the misdetection...

i am the new run-start author

vishwapinnawala, 2020-02-25:
@khassel tried can you try git checkout -b develop --track origin/develop it  
[image]

[image]

vishwapinnawala, 2020-02-25:
@khassel 

[image]

vishwapinnawala, 2020-02-25:
@sdetweil which bash line u want me to add "-x" ??

sdetweil, 2020-02-25:
The 1st line of run-start.sh

What is DISPLAY=:1?

Typically the display on pi is :+1: 0

vishwapinnawala, 2020-02-25:
ahh im on vncserver mode thats why :1

sdetweil, 2020-02-25:
Do u have real display? I don't think this works over vnc

vishwapinnawala, 2020-02-26:
no the previous version worked on my pi with vnc without any error and i tried this version with a monitor too it's the same error but i fixed the issue like @khassel  says in his thread it works for me now

sdetweil, 2020-02-26:
Ok. 

The problem is/was I added code to auto detect when electron could not be used.
Your case I see it starts in server only mode

sdetweil, 2020-02-26:
Can u show me the output of
```
xdpyinfo
```

sdetweil, 2020-02-27:
@vishwapinnawala  how can I recreate this?  I have pi setup to boot and start vnc server. no displays, forced hdmi and display size...

when I vnc in, and logon.. I get the same desktop as with display and DISPLAY=:0.0
not :1 as i your case

using realvnc server  and client. 

if the display is connected, the real display and vnc display are the same. 

when I logon thru vnc (no display connected), pm2 starts MagicMirror, using the run-start.sh on display=:0 just fine..

vishwapinnawala, 2020-02-27:
this is how i do first i setup my pi using a monitor (turning on ssh)and then i remove the monitor and then go headless with ssh using putty in putty i execute "vncserver" command then vncviewer on given ip address in putty

sdetweil, 2020-02-27:
Would really love to see the bash debug output of the run-start.sh.
U can get another copy of it from here
https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/run-start.sh

The stop existing mm. And do

bash -x ~/MagicMirror/run-start.sh > sometime
And send me somefile @ sdetweil at gmail

sdetweil, 2020-02-28:
@vishwapinnawala I might have missed a step, I see some configs that use ssh to create a secure tunnel onto the viewer system, and then use vncviewer  to connect to THAT session.

this also works unchanged.  MagicMirror comes up as before. on display :1

here is the bash trace of that with my comments 
```
+ '[' -z :1.0 ']'                           // if $DISPLAY is not set (is)
++ uname -m
+ arch=armv7l                       //  arm processor type
+ false=false
++ grep -v '^[[:blank:]]*//' config/config.js
++ grep -i serveronly:
++ tr -d ',"'\'''
++ awk '{print tolower($2)}'
+ serveronly=
+ serveronly=false                     // find config.js setting, default false
++ pgrep Xorg                           // check if X is running
+ xorg=539                                      // X is running
++ uname
+ mac=Linux                                   // linux OS
+ '[' false. '!=' false. -o armv7l == armv6l ']'   // if not explicit serveronly -o not armv6 
+ '[' 539. == . -a Linux '!=' Darwin ']'       // if Xwindows runnan d and not Mac
+ node_modules/.bin/electron js/electron.js    // execute electron 

```

closed 2020-05-06 by stale[bot]