MagicMirrorOrg/MagicMirror #1917: MM not working after update

ejmorgan55, 2020-02-03:
Received the following error after update last week. Tried doing both manual and auto update; same error. I cannot find anything about engine.io. Any ideas?

A JavaScript error occurred in the main process
Uncaught Exception:
Error: Cannot find module 'engine.io'
Require stack:
- /home/pi/MagicMirror/node_modules/socket.io/lib/index.js
- /home/pi/MagicMirror/js/server.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/socket.io/lib/index.js:11:14)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)

MichMich, 2020-02-04:
Hi!

- What version of MagicMirror ar you trying to run? `master` or `develop` branch? 
- What steps did you take to perform a manual update?
- Which hardware and which OS are you running?

sdetweil, 2020-02-04:
also, what version of npm and node
npm -v
node -v
and output of uname -a

ejmorgan55, 2020-02-04:
Hi Michael; I updated to the latest version via the web page (MagicMirror Builders); I used the upgrade guide on the site to update; I am running a Raspberry Pi 3b and the latest version of Buster. FYI, after trying to update and getting the errors noted; I ended up deleting the updated MagicMirror folder due to the problems; and used the backup file I had saved. I then re-preformed the update process as prescribed in the upgrade guide and everything began working correctly. I do not know why the prior attempt did not work; albeit, it is possible that there was simply some form of download error causing the engine.io to be deleted. Please close the issue; and thanks for the response.

Ed Morgan

MichMich, 2020-02-04:
Thank for your report!

closed 2020-02-04 by MichMich