MagicMirrorOrg/MagicMirror #1919: [Develop branch] - Electron's white screen during the MM start

sergge1, 2020-02-05:
Do not know whether this is a bug or new electron's behavior.

I just tested MM's dev branch and have found that now the electron's white screen appears just after MM start just for 0.5-1.0 seconds and then reverts to black one.

See video how it looks like: https://youtu.be/Uhb5BjEy-dA

I have never seen such behavior before during the previous installs.

Looks not good with the mirror while everything should be black for good reflection.

khassel, 2020-02-06:
https://github.com/electron/electron/issues/20357

sergge1, 2020-02-07:
Just rolled back to 2.10.1 - and everything works fine, without white screen at electron during the start. 
https://youtu.be/uKVyykPQ4kY

So the same machine, the same environment but **different behavior** in different versions e.g. in **2.11.0** and in **2.10.1**

<details><summary>Environment:</summary><p>

> OS
```js
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
```

> Linux Core
```js
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
```

> Hardware
```js
pi@raspberrypi:~ $ cat /proc/device-tree/model
Raspberry Pi 3 Model B Plus Rev 1.3
```

> Node and NPM versions
```js
pi@raspberrypi:~ $ node -v    v10.18.1
pi@raspberrypi:~ $ npm -v     6.13.4
```

>Electron
```js
pi@raspberrypi:~/MagicMirror $ node_modules/.bin/electron -v
v3.1.13
```
</p></details>

khassel, 2020-02-07:
the problem seems to affect all electron versions >=4. 

So if you use the master branch (v2.10.x, electron v3.x) everything works fine. 

On the develop branch electron was upgraded to 6.x so we have to live with the white screen until this is fixed by electron (or another workaround).

sergge1, 2020-02-08:
so may be it would better to use electron v3.x fi the electron white-screen issue will not be resolved till the MM 2.11.0 release date?
This white screen is really annoying and I expect what we will not lose any specific functionality while using older electron for some time?

MichMich, 2020-02-08:
Buster isn’t supported by the old electron. And if I understand correctly. This is only a issue during startup? Just curious: why would you restart your mirror that often that it is a problem?

sergge1, 2020-02-09:
Yes, issue appears only at startup.
>Buster isn’t supported by the old electron. And if I understand correctly. 
But 2.10.1 works perfect on older Electron version well.
Generally, it is not a problem, I am just perfectionist. If something works visibly wrong, why not to use right/correct one?

My MM restarts few times per day (just to fresh up everything) and I have completely black screen during restart (no 4 raspberries at start, no rainbow screen at start, no splash screen, no colored desktop; so such white blink looks not so good for me. 
But it is not crucial. Just raised issue because tested DEV branch and pointed out this issue.

MichMich, 2020-02-09:
Just curious: Why do you need to restart/refresh your mirror? My mirror restarts once every 3 months: when I run the update. Never have any issues.

dfanica, 2023-01-03:
I would like to see if there's a potential fix for this issue as well.
As @sergge1 said, it's not an actual issue, just an annoying behaviour. I also have a small piece of tape I've put in the corner of my mirror that is visible with the white background :)) just reminds me of my sloppy work!

If someone figures out how to change that to a black color instead of white that would be awesome ;)

rejas, 2023-01-03:
Are you running on the latest MM version with all packages updated?

dfanica, 2023-01-03:
> Are you running on the latest MM version with all packages updated?

yes, I just updated to 2.22 right before looking into this issue... which was present before the update as well

sdetweil, 2023-01-03:
i've been trying to get rid of the white flash for years.. without success..    depends on system speed really..

dfanica, 2023-01-03:
> i've been trying to get rid of the white flash for years.. without success.. depends on system speed really..

I'm determined to get rid of this TODAY!

sdetweil, 2023-01-03:
i do not see that error.. 

I have tried adding code to force the initial background, minimize and then restore after ready, ... 
nothing has helped..  I have seen it on every mirror system I have ever built..
pi, 0w, 02w, pi3, pi4, odroid xu4, jetson nano, mac vm, mac mini, linux x86, gallium,  ubuntu

dfanica, 2023-01-03:
The final solution to this works on Raspberry Pi4, where I've been testing and have installed MagicMirror.

The issue is caused by the electron option `fullscreen: true`.
Following the article posted by @sdetweil, https://www.christianengvall.se/electron-white-screen-app-startup/, I'm using a similar approach.
The idea is to avoid using the `fullscreen` approach to this which flashed the white screen upon load.

First step would be to comment out the fullscreen option being set. Look for the line below and comment it out.
```
//electronOptionsDefaults.fullscreen = true;
```

Second step would be to add the following extra option, potentially replacing the fullscreen one from step one.
```
electronOptionsDefaults.frame = false;
```

Adding the following at the end of the `createWindow` method is the third step:
```
	mainWindow.once('ready-to-show', () => {
		mainWindow.maximize();
		mainWindow.show();
 	});
```
This essentially removes the fullscreen option, waits for the app to be ready to render and then displays the content while maximizing the window. As long as `frame: false` is set you will not see a title bar and everything should look smooth.

I will be looking into creating a PR to fix this issue in this way but I need to test more to make sure this approach doesn't break other implementations.

sdetweil, 2023-01-03:
cool.. that never worked before!!

dfanica, 2023-01-03:
> cool.. that never worked before!!

I told you I'm determined to find a workaround for this issue TODAY! I hope it works for you as well... and maybe other platforms.

rejas, 2023-01-04:
Might this be something worth adding to the documentation : 
https://docs.magicmirror.builders/configuration/raspberry.html 
?

sdetweil, 2023-01-04:
it happens on other platforms too..  maybe his fix works

khassel, 2023-01-04:
tested this on my rpi4, the only thing I needed to avoid the white screen is to add `electronOptionsDefaults.transparent = true;`

@rejas @sdetweil can you test this with your setup?

closed 2023-04-04 by khassel