MagicMirrorOrg/MagicMirror #1920: MagicMirror2 only shows black screen no modules, no errors at all

android9pie2, 2020-02-06:
MagicMirror2 starts up, only shows a black screen. There are no errors in the log, and I have my config file below. I have tried ```npm serveronly``` ,same thing just in my browser. I am running a Raspberry Pi 4b with Raspbian 10 using the Budgie Desktop. I cannot afford to reinstall right now, because then I would have to reinstall all my modules and I don't have time, so I would appreciate every possible solution besides re-installation. 

Modules
```
MMM-Cursor
MMM-AssistantMk2
WallberryTheme
MMM-Hotword
```
config.js
```
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "", "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	language: "en",
	timeFormat: 12,
	units: "imperial",

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "CNet",
						url: "http://news.cnet.com/2547-1_3-0-20.xml"
					},
					{
						title: "Android Authority",
						url: "http://www.androidauthority.com/feed/"
					},
{
						title: "Engadget",
						url: "https://www.engadget.com/rss.xml"

					},
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
{
	module: "MMM-Hotword",
	position: "top_left",
	config: {
		chimeOnFinish: false,
		mic: {
			recordProgram: "arecord",
			device: "plughw:1"
		},
		models: [
			{
				hotwords    : "smart_mirror",
				file        : "smart_mirror.umdl",
				sensitivity : "0.9",
			},
		],
		commands: {
			"smart_mirror": {
				notificationExec: {
					notification: "ASSISTANT_ACTIVATE",
					payload: (detected, afterRecord) => {
						return {profile:"default"}
					}
				},
				restart:false,
				afterRecordLimit:0
			}
		}
	}
},
{
  module: "MMM-AssistantMk2",
  position: "top_left",
  config: {
    ui: "classic2", // or "Classic", "Classic2"
    assistantConfig: {
      projectId: "integration-service-6339c", // Required to use gaction.
      modelId: "integration-service-6339c-mirrorx-v8kg0x", // (OPTIONAL for gaction)
      instanceId: "harmony-home-mirrorx", // (OPTIONAL for gaction)
      latitude: 51.508530,
      longitude: -0.076132,
    },
    responseConfig: {
      useHTML5: true, // sound render by HTML5
      useScreenOutput: true,
      useAudioOutput: true,
      useChime: true,
      timer: 5000,
      myMagicWord: false,
      //Your prefer sound play program. By example, if you are running this on OSX, `afplay` could be available.
      playProgram: "mpg321", //needed if you don't use HTML5 sound render
      chime: {
        beep: "beep.mp3",
        error: "error.mp3",
        continue: "continue.mp3",
        open: "Google_beep_open.mp3",
        close: "Google_beep_close.mp3",
      },
      // false - animated icons, 'standby' - static icons only for standby state, true - all static icons
      useStaticIcons: false
    },
    micConfig: { // put there configuration generated by auto-installer
      recorder: "arecord",
      device: "plughw:1",
    },
    customActionConfig: {
      autoMakeAction: false,
      autoUpdateAction: false, // in RPI, gaction CLI might have some trouble.(current version should be 2.2.4, but for linux-arm, Google haven't updated) so leave this as false in RPI. I don't know it is solved or not.
      actionLocale: "en-US", // At this moment, multi-languages are not supported, sorry. Someday I'll work.
    },
    recipes: [ "with-MMM-Hotword.js", "with-MMM-TelegramBot.js" ],
    profiles: {
      "default": {
        profileFile: "default.json",
        lang: "en-US"
      }
    },
    addonsConfig: {}, //describe addons configuration
  },
},
{
    module: "WallberryTheme",
    position: "fullscreen_below", // Required Position
    config: {
      unsplashAccessKey: "5f17b902e9bebcb24c9a5e2282a9223569fdd2286a9bfeae14e83e280ba0d63b", // REQUIRED
      collections: "343012" // optional - leave empty for a random photo
    }
  },
  // WB-clock adds local time (Optional Module)
  {
    module: "WallberryTheme/WB-clock",
    position: "top_bar", // highly suggest using top_bar position
    config: {
      localCityName: "Hello Naaman", // optional
      
    }
  },
  // WB-weather adds weather via DarkSky API (Optional Module)
  {
    module: "WallberryTheme/WB-weather",
    position: "bottom_bar",  // Highly suggested location
    config: {
      // See "Configuration options" for more information.
      daysToForecast: "3",
      darkSkyApiKey: "ff278ad56d6f9550f05c22862ebc6fed", // REQUIRED
      latitude:   29.558123483125385, // REQUIRED
      longitude: -95.41360222750056 // REQUIRED

    }
},
 

{
		module: 'MMM-Cursor',
		config: {
			// See 'Configuration options' for more information.
		}
	},

	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
```
Please help me.

android9pie2, 2020-02-06:
Ok, I figured it out from a little debugging I did myself. It is the Google Assistant module, and it isn't something I have the time to try and fix myself. So far, the developer hasn't even answered the questions of others who are having the same problem.

sdetweil, 2020-02-18:
@android9pie2   can you close this issue now?

closed 2020-04-25 by stale[bot]