MagicMirrorOrg/MagicMirror #1922: Default Calendar fails to load as of 2/7/20

bdin1, 2020-02-09:
0|mm     | Create new calendar fetcher for url: http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics - Interval: 300000
0|mm     | [17:52:42.010] [LOG]
0|mm     | Create new calendar fetcher for url: https://calendar.google.com/calendar/ical<my google calendar>/private-bbf487ef0035dfe5709683aa2738a1c5/basic.ics - Interval: 300000
0|mm     | [17:52:42.121] [LOG]
0|mm     | Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
0|mm     | [17:52:43.613] [LOG]
0|mm     | Whoops! There was an uncaught exception...
0|mm     | [17:52:43.614] [ERROR]
0|mm     | Error: Unknown RRULE property ''
0|mm     |     at /home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1207:23
0|mm     |     at Array.forEach (<anonymous>)
0|mm     |     at parseRrule (/home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1165:11)
0|mm     |     at parseLine (/home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1148:16)
0|mm     |     at Array.map (<anonymous>)
0|mm     |     at Function.parseString (/home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1126:41)
0|mm     |     at Function.RRule.fromString (/home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:2150:32)
0|mm     |     at Object.ical.objectHandlers.END (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:62:23)
0|mm     |     at Object.handleObject (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:400:41)
0|mm     |     at Object.parseICS (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:441:20)
0|mm     | [17:52:43.624] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|mm     | [17:52:43.625] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues

dtreskunov, 2020-02-10:
I noticed this as well. Opened https://github.com/peterbraden/ical.js/issues/109 on the ical library.

sdetweil, 2020-02-10:
this is actually a problem with the calendar data. it generates empty RRULE: statements..

the rrule nodejs library cannot handle this

edit the rrule/dist/es5/rrule.js file and insert the line below
```
            case 'BYEASTER':
                options.byeaster = Number(value);
                break;
            case '':                      // insert
                break;                  // insert
            default:
```

note also the calendar ics file has duplicates of all events..  so the file is very messed up

bdin1, 2020-02-14:
This worked, thank you.

sdetweil, 2020-02-18:
calendarlabs reported that the empty RRULE problem has been fixed.  I don't see any more failures on new installs...

sdetweil, 2020-04-05:
we can close this now, was calendar labs formatting problem

bdin1, 2020-04-05:
Okay, I thought it was closed

Eric Carrozzo
318-729-9912

closed 2020-04-06 by MichMich