MagicMirrorOrg/MagicMirror #1926: default calendar needs to use ical as a dependency vs self embedded library.   we are woefully behind

sdetweil, 2020-02-10:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

MichMich, 2020-02-10:
We have used it as a dependency before, but that gave a lot of issues back then. Might be a good idea to give it another try, though!

sdetweil, 2020-02-10:
see #1922

mdissel, 2020-04-17:
I've got an issue with a yearly repeated event that is shown in the calendar as a monthly recurring event.. Seems related to the mentioned issues

rossburton, 2020-06-16:
Please do this, there's lots of issues with the calendar.

rejas, 2020-06-30:
Should be closable since https://github.com/MichMich/MagicMirror/pull/2048 got merged @sdetweil ?

closed 2020-07-01 by sdetweil