MagicMirrorOrg/MagicMirror #1931: stopping MM

erkilic, 2020-02-15:
Hi,
I have a question regarding stopping MagicMirror via shell or terminal command. So if I have started MM with "npm start" is there also a possibility to stop it with a command (not only with shortcut). 
Rational: I want to stop working MM with a Python code.

Thx in advance.
Gürkan

sdetweil, 2020-02-16:
i wrote a script to do that..

needs the moreutils package installed

sudo apt-get install moreutils

I call this 
clear_node
have to make it executable 
chmod +x clear_node
```
#!/bin/bash

ps -ef | grep pocket | grep -v "grep"  | awk '{print $2}'  | ifne xargs sudo kill -9
ps -ef | grep node | grep -v "grep"  | awk '{print $2}' | ifne xargs sudo kill -9
ps -ef | grep chromium-browser | grep -v "grep"  | awk '{print $2}' | ifne xargs sudo kill -9
sudo killall  chromium-browser
sudo killall  /usr/lib/chromium-browser/chromium-browser
sudo killall electron
```

erkilic, 2020-02-16:
thanks a lot MichMich. And thanks for your amazing MagicMirror Project! 

⁣BlueMail for Android herunterladen ​

sdetweil, 2020-04-05:
this can be closed

closed 2020-04-06 by MichMich