MagicMirrorOrg/MagicMirror #1932: [19:11:01.185] [LOG]    Whoops! There was an uncaught exception... [19:11:01.187] [ERROR]  Error: Unknown RRULE property ''

huynhquoctoan, 2020-02-16:
[19:11:01.185] [LOG]    Whoops! There was an uncaught exception...
[19:11:01.187] [ERROR]  Error: Unknown RRULE property ''
    at /home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1191:23
    at Array.forEach (<anonymous>)
    at parseRrule (/home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1149:11)
    at parseLine (/home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1132:16)
    at Array.map (<anonymous>)
    at Function.parseString (/home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1110:41)
    at Function.RRule.fromString (/home/pi/MagicMirror/node_modules/rrule/dist/es5/rrule.js:2135:32)
    at Object.ical.objectHandlers.END (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:62:23)
    at Object.handleObject (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:400:41)
    at Object.parseICS (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:441:20)
[19:11:01.203] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[19:11:01.204] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[19:11:41.531] [LOG]    Shutting down server...
[19:11:41.534] [LOG]    Stopping module helper: updatenotification
[19:11:41.536] [LOG]    Stopping module helper: calendar
[19:11:41.538] [LOG]    Stopping module helper: newsfeed
[19:11:41.539] [LOG]    Stopping module helper: MMM-AssistantMk2
pi@raspberrypi:~/MagicMirror $

mangoman18, 2020-02-16:
Can confirm this error. It is showing up on my RasberryPI as well

khassel, 2020-02-16:
duplicate of #1922

sdetweil, 2020-02-18:
this has been resolved by the calendarlabs.com folks.

sdetweil, 2020-04-05:
this can be closed

closed 2020-04-06 by MichMich