MagicMirrorOrg/MagicMirror #1935: MM zoom control

DennisEJ, 2020-02-17:
I added zoom to my config.js file of 1.5.  It increased the size of my calendar where all the items are overlaying the others.  I went back and changed zoom to 1 (the default) and started it again and the size was still overlapping.  I restarted MM and it is still overlapping.  I removed zoom all together and that didn't help.  I don't know how to get back to where the normal size of the calendar items are not overlapping.Please only submit reproducible issues. 

I can't find a reference to fixing this problem.
Any help?
Thanks
DennisEJ

sdetweil, 2020-02-18:
use the ctrl and minus keys to zoom out

DennisEJ, 2020-02-18:
Thank you... must be terrible having to answer questions for newbies....
DennisEJ

closed 2020-02-18 by DennisEJ

sdetweil, 2020-02-18:
@DennisEJ  no.. i was a newbie not so long ago and still am on lots of things..

this is the same as on browsers on Windows..  ctrl+ to zoom in and ctrl- to zoom out

hangorazvan, 2020-03-09:
zoom in config is only for electron