MagicMirrorOrg/MagicMirror #1950: MagicMirror.NET (Windows version based on WPF)

jdahan91, 2020-03-10:
Hi, yesterday I started a port of an existing system in WinForms to WPF and I think it has the potential to become a modular system for Windows-based systems which has similar capabilities as the one written in js. 

The thing is, the current one is only for linux based systems which is limited. My solution is to write one for Windows and even make it (in the future) cross platform. Of course the communication will be based on Telegram (or any other interface) and will have many features written as plugins either by me or individuals.

My question is, is it ok for me to use the name "MagicMirror".NET as a Windows based port to the system? Thanks.

MichMich, 2020-03-10:
Thanks for reaching out. It can run on windows without any problems. That being said, I prefer to not use MagicMirror as project name to prevent any confusion.

jdahan91, 2020-03-10:
> Thanks for reaching out. It can run on windows without any problems. That being said, I prefer to not use MagicMirror as project name to prevent any confusion.

Alright, I will rename the solution/project.

closed 2020-03-10 by jdahan91