MagicMirrorOrg/MagicMirror #1956: weird .resume() behaviors : executed regargless of context

eouia, 2020-03-19:
**Platform**: All

**Node Version**: no matter

**MagicMirror Version**: at least 2.10 over

**Description**: `.resume()`(and `.suspend()`) is executed whenever `.show()` is called regardless of context. It might not be a bug but useless in real usage.
- executable even when the module is already revealed.
- executable with wrong `lockString`, so the module itself should not be revealed really, but `.resume()` is called.
- executable under multiple locked, so the module itself should not be revealed really, but `.resume()` is called.

Inside of `.resume()`, I can't find any way to measure this calling is proper on not. `.lockStrings` and `.hidden` would be changed after this `.resume()` calling. 

Description of dev guide says. 
```
When a module is requested to be shown (using the module.show() method), the resume() method will be called. 
By subclassing this method you can perform tasks restarting the update timers.
```
But by subclassing, I cannot perform tasks restarting properly. Because whenever any module calls `.show()`, `.hide()`,  `.resume()` and `.suspend()` would be called without context, so my task would become a mess.

Is it the right behavior? If so, this `.resume()` looks useless except for counting how many times `.show()` is called. 
Or is there a way to determine a proper calling inside of `.resume()`? 

**Steps to Reproduce**: List the step by step process to reproduce the issue.
I tested with these test modules.

**module:test1**
```js
Module.register("test1", {
  start: function(){
    this.showing = true
  },

  suspend: function () {
    console.log("@stop", this.lockStrings, this.hidden)
    this.showing = false
  },

  resume: function() {
    console.log("@resume", this.lockStrings, this.hidden)
    this.showing = true
  },

  notificationReceived: function() {

  },
})
```

**module:test2**
```js
Module.register("test2", {
  notificationReceived: function(noti, payload, sender) {
    if (noti == "DOM_OBJECTS_CREATED") {
      this.test()
    }
  },

  test: function() {
    const asleep = async (ms) => {
      return new Promise(resolve => setTimeout(resolve, ms))
    }

    var target = null

    MM.getModules().enumerate((m)=>{
      if (m.name == "test1") target = m
    })

    const scenario = async () => {
      await asleep(1000)
      console.log("1. Try to show when target is revealed already")
      target.show()
      await asleep(1000)
      console.log("2. Try to hide with lockString 'abcd'")
      target.hide(0, {lockString:'abcd'})
      await asleep(1000)
      console.log("3. Try to show with wrong lockString '1234'")
      target.show(0, {lockString:'1234'})
      await asleep(1000)
      console.log("4. Try to show with right lockString 'abcd'")
      target.show(0, {lockString:'abcd'})
      await asleep(1000)
      console.log("5. Try to hide twice (lockString 'a', 'b')")
      target.hide(0, {lockString:'a'})
      await asleep(500)
      target.hide(0, {lockString:'b'})
      await asleep(1000)
      console.log("6. Try to unlock once. module will not be shown.")
      target.show(0, {lockString:'b'})
      await asleep(1000)
      console.log("7. Try to unlock once more. now module will be shown.")
      target.show(0, {lockString:'a'})
    }

    scenario()

  }
})
```

**Expected Results**: 
I expect `.resume()` would be called on scenario no. 4 and 7. (hmmm, no.1 also be acceptable at least)
**Actual Results**: 
[image]

sdetweil, 2020-03-19:
really this is a bug in module.js, it calls resume, before checking if it should

```
	show: function (speed, callback, options) {
		if (typeof callback === "object") {
			options = callback;
			callback = function () { };
		}

		callback = callback || function () { };
		options = options || {};

		this.resume();   // < ---  should not be here
		MM.showModule(this, speed, callback, options);
	}
```

bugsounet, 2020-03-19:
we do some test,
it's not the real place for this.resume()
It should be in the before the callback
I make some patch and test I better like this ;)

and now it's works !

closed 2020-03-20 by eouia